liudf0716/apfree-wifidog #24: error while compiling package

Tr4cer0ute, 2017-02-03:
Hi,

I got this error while compiling IPK

```
make[3]: Leaving directory `/Volumes/OpenWRT/openwrt-CC/package/kernel/mac80211'
make[2]: Leaving directory `/Volumes/OpenWRT/openwrt-CC'
make[1]: *** [/Volumes/OpenWRT/openwrt-CC/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.package_compile] Error 2

```

I'm using MAC OS X as a compilation environment

liudf0716, 2017-02-06:
I think this error information can not make me find where your problem, can u give me more info ?

liudf0716, 2017-02-06:
by the way, what u provide information seems unrelated with apfree wifidog project.

Tr4cer0ute, 2017-02-06:
@liudf0716 Should I post the whole log file? Why it's unrelated? I followed up the normal process to compile the package under Openwrt CC

liudf0716, 2017-02-07:
how u compile？ for example, in my openwrt/LEDE environment,   I use such command:
make package/apfree/apfree_wifidog/compile V=s

Tr4cer0ute, 2017-02-07:
Same here. This is not the first time I am compiling a package under Openwrt. An other error prompt 

```
Makefile:12: ../../luci.mk: No such file or directory
make[4]: *** No rule to make target `../../luci.mk'.  Stop.
```

liudf0716, 2017-02-08:
please do like this:

Fork and clone the Apfree WiFiDog project:

git clone https://github.com/liudf0716/apfree_wifidog
cd apfree_wifidog
Assuming you have a working LEDE/Openwrt setup, taking LEDE as an example and assuming your LEDE root path is LEDE_ROOT:

cp -r package/apfree_wifidog/ /LEDE_ROOT/package/
To support HTTPS, you need install libevent with version 2.1.7 or latest in your LEDE environment, Or using the package copied in Apfree WiFiDog git project:

cp -r package/libevent2/ /LEDE_ROOT/package/libs/
Now Apfree WiFiDog package has been installed in LEDE packages environment.

cd /LEDE_ROOT/
make menuconfig
Chose your Target System and ApFree --> apfree_wifidog. SAVE and EXIT.

make package/apfree_wifidog/compile V=s

Tr4cer0ute, 2017-02-11:
.files-packageinfo.mk:3: *** target pattern contains no `%'.  Stop.
make: *** [prepare-tmpinfo] Error 2

liudf0716, 2017-02-13:
too few information, could u provide more information?

Tr4cer0ute, 2017-02-13:
@liudf0716 This is too weird...I will revert with much details as I can.

liudf0716, 2017-02-13:
ok. I also think it's wired, nobody encountered such problem

Tr4cer0ute, 2017-02-21:
Linking C executable wifidog
CMakeFiles/wifidog.dir/https_server.c.o: In function `thread_https_server':
/Volumes/OpenWRT/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/apfree_wifidog/apfree_wifidog-1.12.797/src/https_server.c:(.text+0x42a): undefined reference to `evhttp_set_bevcb'
collect2: error: ld returned 1 exit status
make[5]: *** [src/wifidog] Error 1
make[4]: *** [src/CMakeFiles/wifidog.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [/Volumes/OpenWRT/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/apfree_wifidog/apfree_wifidog-1.12.797/.built] Error 2
make[2]: Leaving directory `/Volumes/OpenWRT/openwrt/package/apfree_wifidog'
make[1]: *** [package/apfree_wifidog/compile] Error 2
make[1]: Leaving directory `/Volumes/OpenWRT/openwrt'
make: *** [package/apfree_wifidog/compile] Error 2

liudf0716, 2017-02-22:
@Tr4cer0ute u need update your libevent to latest version 2.1.8-stable
the following is my PR for LEDE
https://github.com/lede-project/source/pull/848

Tr4cer0ute, 2017-02-24:
Finally! did nothing magic

closed 2017-02-24 by Tr4cer0ute

liudf0716, 2017-02-25:
@Tr4cer0ute compile passed?

Tr4cer0ute, 2017-02-25:
yes. working with ur source repo

liudf0716, 2017-02-25:
Good

gigibox, 2017-02-25:
very good！😄