liudf0716/apfree-wifidog #30: Wifidog auth server

Tr4cer0ute, 2017-02-28:
Hi,

I have two concerns right here after successfuly compiling Apfree.

1. Can I use it with my actual auth server? 
2. How can I put my own gateway ID? /tmp/wifidog.conf lost values after restarting

Best

liudf0716, 2017-03-01:
@Tr4cer0ute 
1, yes, of course u can use your own auth server , only if your server support wifidog protocol. In fact , there are tens of thousands openwrt devices used apfree wifidog.
2,  I used br-lan mac address for gateway ID,  I suggest u used it too, but if u really want to use your own gateway ID, u can change it in wifidog.init file, which get gateway_id like this : 
 local default_gateway_id=`ifconfig br-lan|grep HWaddr|awk '{print $5}'|sed s/://g`
u can change it as you wish

liudf0716, 2017-03-01:
@Tr4cer0ute  I have submitted my project as pr for openwrt package, can u give me some encouragement in my PR? here is my PR:
https://github.com/openwrt/packages/pull/4012

Tr4cer0ute, 2017-03-01:
@liudf0716 I changed it but still prompt that auth server is offline. I have an other device connected to the same auth server (authpuppy).

I will be more than glad to support your project.

Tr4cer0ute, 2017-03-01:
it's oky, I put 80 instead of 2060 for auth server default port and it works like a charm!

closed 2017-03-01 by Tr4cer0ute