liudf0716/apfree-wifidog #34: Captive portal behavior

Tr4cer0ute, 2017-03-02:
Hi,

After two days of Apfree testing I noticed two issues:

1- Relative to android devices, each time my phone connect to WiFi a certificate error is prompting asking for confirmation in order to proceed.

2 - Apple devices doesn't automatically prompt the captive portal, I think apple domain names are allowed by default.

What do you think of this?

Tr4cer0ute, 2017-03-02:
For the first point I think that your daemon is integrating a self signed certificate. Can we put our own?

liudf0716, 2017-03-02:
@Tr4cer0ute 
1- Relative to android devices, each time my phone connect to WiFi a certificate error is prompting asking for confirmation in order to proceed.

this problem was caused by ssl or tls, when client find server's cert is illegal, it will prompt, some browser even stop the connection. this problem is unresolvable, if it can be resolved, it will turn out ssl or tls is unsafe.

2 - Apple devices doesn't automatically prompt the captive portal, I think apple domain names are allowed by default.

I don't know why your ios device can not prompt automatically, we tested it's ok. maybe in your country, apple did not use captive.apple.com for its wisper?

Tr4cer0ute, 2017-03-02:
@liudf0716 

1 - what if we put a valid ssl cert? 
2 - Absolutely no, it's captive.apple.com

I notice that when all ping test succeed event if I am not authorized, based on my knowledge only DNS queries are allowed behind a captive unless the user is authorized

liudf0716, 2017-03-02:
@Tr4cer0ute 
1 - what if we put a valid ssl cert?
what u mean valid ssl cert? if u refered a valid ssl cert as cert signed by legal CA center, the same result with youself signed cert. only if u get what you visit web's cert, for example, if u visit google, u get google's cert and key file, but that is impossible.

2 - Absolutely no, it's captive.apple.com
Ok, I know why you have no prompt page. cause in apfree wifidog, when it  receive second captive.apple.com visit for more than 20 seconds, it will automatically add this device into temporary white mac list for a minute.
u can wait for over a minute, then u will be blocked.

closed 2017-03-19 by liudf0716