liudf0716/apfree-wifidog #45: lede 17.01 make 失败

nu1ee, 2017-03-11:
src/CMakeFiles/wifidog.dir/build.make:710: recipe for target 'src/CMakeFiles/wifidog.dir/mqtt_thread.c.o' failed
make[5]: *** [src/CMakeFiles/wifidog.dir/mqtt_thread.c.o] Error 1
make[5]: Leaving directory '/home/nulee/work/lede-17.01/build_dir/target-x86_64_musl-1.1.16/apfree_wifidog/apfree_wifidog-2.03.1061'
CMakeFiles/Makefile2:124: recipe for target 'src/CMakeFiles/wifidog.dir/all' failed
make[4]: *** [src/CMakeFiles/wifidog.dir/all] Error 2
make[4]: Leaving directory '/home/nulee/work/lede-17.01/build_dir/target-x86_64_musl-1.1.16/apfree_wifidog/apfree_wifidog-2.03.1061'
Makefile:127: recipe for target 'all' failed
make[3]: *** [all] Error 2
make[3]: Leaving directory '/home/nulee/work/lede-17.01/build_dir/target-x86_64_musl-1.1.16/apfree_wifidog/apfree_wifidog-2.03.1061'
Makefile:67: recipe for target '/home/nulee/work/lede-17.01/build_dir/target-x86_64_musl-1.1.16/apfree_wifidog/apfree_wifidog-2.03.1061/.built' failed
make[2]: *** [/home/nulee/work/lede-17.01/build_dir/target-x86_64_musl-1.1.16/apfree_wifidog/apfree_wifidog-2.03.1061/.built] Error 2
make[2]: Leaving directory '/home/nulee/work/lede-17.01/package/ext/apfree_wifidog'
package/Makefile:105: recipe for target 'package/ext/apfree_wifidog/compile' failed
make[1]: *** [package/ext/apfree_wifidog/compile] Error 2
make[1]: Leaving directory '/home/nulee/work/lede-17.01'
/home/nulee/work/lede-17.01/include/toplevel.mk:197: recipe for target 'package/ext/apfree_wifidog/compile' failed
make: *** [package/ext/apfree_wifidog/compile] Error 2
这个怎么处理啊  依赖包 已经更换

liudf0716, 2017-03-13:
can u provide me more information? do u add depends on libmosquitto?

nu1ee, 2017-03-13:
[build.txt](https://github.com/liudf0716/apfree_wifidog/files/837878/build.txt)

nu1ee, 2017-03-13:
分析出来 是这个无效
// liudf added 20160411
extern pthread_mutex_t domains_mutex;

error: unknown type name 'pthread_mutex_t'

nu1ee, 2017-03-13:
版本2.03.1061 已修复此问题 
还是编译失败，后来删除源码包重下载 就好了

closed 2017-03-13 by nu1ee