liudf0716/apfree-wifidog #57: Apfree stop working when I put trusted domains

Tr4cer0ute, 2017-03-29:
Hi,

Apfree stop working and connect people automatically when I set Facebook trusted domains

here my line

option trusted_domains 'facebook.com,fbcdn.net,facebook.net,web.facebook.com,akamaihd.net'

liudf0716, 2017-03-30:
can u show me file '/tmp/wifidog.conf', that file was real used by apfree wifidog.
by the way, if u want user can access facebook, u can try pan domain, like this:
wdctl add_trusted_pdomains facebook.com,fbcdn.net,facebook.net,akamaihd.net
if u want to save what u added rules, u can use 
'wdctl user_cfg_save'

Tr4cer0ute, 2017-03-30:
Walled garden is not working properly, FBCDN.net subdomain are blocked (can't get ressources)

liudf0716, 2017-03-31:
@Tr4cer0ute u should use pan domains. I think u want all *.fbcdn.net be accessed. but trusted domain only resolve fbcdn.net's ip, which should not be what u want.

Tr4cer0ute, 2017-03-31:
@liudf0716 That's it. I want to allow all sub domains. What is the option line under configuration file? is it 

> option trusted_pan_domains

Tr4cer0ute, 2017-04-04:
@liudf0716 please point to the required option

closed 2017-04-04 by Tr4cer0ute