liudf0716/apfree-wifidog #60: Failed to build for MIPS architecture

Tr4cer0ute, 2017-04-04:
Hi, 

I'am trying to compile apfree for MIPS architecture but it won't success, knowing that I'm using the same build to compile ar71x and it working fine.

Tr4cer0ute, 2017-04-04:
Here is the error

> Volumes/OpenWRT/openwrt-archer-c2/staging_dir/toolchain-mipsel_24kec+dsp_gcc-5.3.0_musl-1.1.14/lib/gcc/mipsel-openwrt-linux-musl/5.3.0/../../../../mipsel-openwrt-linux-musl/bin/ld: copy reloc against protected `__stack_chk_guard' is dangerous
> CMakeFiles/wifidog.dir/https_server.c.o: In function `serve_some_http':
> /Volumes/OpenWRT/openwrt-archer-c2/build_dir/target-mipsel_24kec+dsp_musl-1.1.14/apfree_wifidog/apfree_wifidog-1.12.797/src/https_server.c:313: undefined reference to `evhttp_set_bevcb'
> collect2: error: ld returned 1 exit status
>

liudf0716, 2017-04-05:
u need use libevent2 2.1.8 version

closed 2017-04-07 by liudf0716