liudf0716/apfree-wifidog #73: Error loading shared library libevent-2.1.so.6: No such file or directory (needed by /usr/bin/wifidog)

Tr4cer0ute, 2017-04-20:
got this error while starting WiFidog

Error loading shared library libevent-2.1.so.6: No such file or directory (needed by /usr/bin/wifidog)
Error loading shared library libevent_openssl-2.1.so.6: No such file or directory (needed by /usr/bin/wifidog)

Package installation process was normal

liudf0716, 2017-04-21:
you should install that two libevent package.

Tr4cer0ute, 2017-04-21:
They are already installed.

liudf0716, 2017-04-21:
maybe you install is libevent-2.0**,  
ls /usr/lib/

Tr4cer0ute, 2017-04-21:
iptables              libip6tc.so.0         libnl-tiny.so
libhttpd.so           libip6tc.so.0.1.0     libuclient.so
libhttpd.so.0         libiptext.so          libxtables.so
libhttpd.so.0.0.0     libiptext4.so         libxtables.so.10
libip4tc.so           libiptext6.so         libxtables.so.10.0.0
libip4tc.so.0         libiwinfo.so          opkg
libip4tc.so.0.1.0     libjson-c.so.2        os-release
libip6tc.so           libjson-c.so.2.0.2    pppd

liudf0716, 2017-04-22:
seems there are not libevent library

Tr4cer0ute, 2017-04-29:
I compiled bin image with apfree and libevent2 library... I will work on this during this weekend. I will revert results.

Tr4cer0ute, 2017-04-29:
it's oky I recompiled using libevent2 and its working now

closed 2017-04-29 by Tr4cer0ute

Lotus253, 2017-09-06:
sudo ln -s /usr/lib/libevent-2.1.so.6  /usr/lib64/libevent-2.1.so.6

sasanbozorg, 2018-03-06:
please help

telegram-cli: error while loading shared libraries: libevent-2.0.so.5: cannot open shared object file: No such file or directory