liudf0716/apfree-wifidog #86: openwrt 15.05.1 do not compile if I add appfree wifidog

JJEU, 2017-04-28:
Hello, 

I successfully compiled openWRT chaos Calmer 15.05.1 today, but if i add apfree_wifidog, it doesn't compile

here is the full list of commands :
```
git clone -b chaos_calmer git://github.com/openwrt/openwrt.git
 cd  <buildroot>
./scripts/feeds update -a
./scripts/feeds install -a
```

in another folder : 
```
git clone https://github.com/liudf0716/apfree_wifidog
cd apfree_wifidog
cp -r package/apfree_wifidog/ /<buildroot>/package/
```

I did not used you libevent2 package since I already selected the libevent2 package included with openWRT

`make menuconfig`

selected the following packages : 

```
Target System: Atheros (AR7xxx/AR9xxx)
    Subtarget: Generic
    Target Profile : Ubiquiti products (note : il y a un profil pour Unifi, à tester)
    Target image : squashfs
Luci
	Collections > Luci, Luci-ssl 
	applications luci-app-diag-core, luci-app-commands, , luci-app-ntpc,  luci-app-sqm, luci-app-vnstat,luci-app-watchcat , luci-app-ddns, luci-apfree-wifidog
Administration
	htop
Utilities
	Editors > Nano 
	watchcat
Network
	Captive portals > apfree_wifidog
	File transfert > curl
	SSH > openssh-sftp-server
	iperf3
Languages
	Perl > Perl
Administration
	htop
Libraries
	SSL : libopenSSL
	libevent2
```
launched a download first, two times to verify in the log that everything is ok with the repositories server : 
`make download V=s 2>&1 | tee log-makedownload.txt | grep -i error`

then i compiled : 
`sudo nice -n 18 make V=s 2>&1 | tee log-make-datedujour.txt | grep -i error`

the log is very long. i add it to this post. it ends with 
```
Makefile:67: recipe for target '/media/zap/openwrtdisk/chaosCalmer2017/AirRouter/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/apfree_wifidog/apfree_wifidog-2.03.1061/.built' failed
make[3]: *** [/media/zap/openwrtdisk/chaosCalmer2017/AirRouter/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/apfree_wifidog/apfree_wifidog-2.03.1061/.built] Error 2
make[3]: Leaving directory '/media/zap/openwrtdisk/chaosCalmer2017/AirRouter/openwrt/package/apfree_wifidog'
package/Makefile:191: recipe for target 'package/apfree_wifidog/compile' failed
make[2]: *** [package/apfree_wifidog/compile] Error 2
make[2]: Leaving directory '/media/zap/openwrtdisk/chaosCalmer2017/AirRouter/openwrt'
package/Makefile:188: recipe for target '/media/zap/openwrtdisk/chaosCalmer2017/AirRouter/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.package_compile' failed
make[1]: *** [/media/zap/openwrtdisk/chaosCalmer2017/AirRouter/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory '/media/zap/openwrtdisk/chaosCalmer2017/AirRouter/openwrt'
/media/zap/openwrtdisk/chaosCalmer2017/AirRouter/openwrt/include/toplevel.mk:181: recipe for target 'world' failed
make: *** [world] Error 2
```

i attach the .config file and the compile log file to this post

thanks !
[config.txt](https://github.com/liudf0716/apfree_wifidog/files/963379/config.txt)
[log-make-27april2017.txt](https://github.com/liudf0716/apfree_wifidog/files/963380/log-make-27april2017.txt)

liudf0716, 2017-04-28:
@JJEU please use our libevent2 package, cause apfree wifidog need latest libevent version to support its https redirect. I have submitted pull request to update libevent2 version in LEDE project long time ago, but unfortunately they still not accepted my pull request.

liudf0716, 2017-04-28:
I can not read what u attached file, can u provide output of the following command?
make package/apfree_wifidog/compile V=s

closed 2017-06-14 by liudf0716