liudf0716/apfree-wifidog #95: wdctl add_trusted_pdomains

Tr4cer0ute, 2017-04-30:
wdctl add_trusted_pdomains [domain1.com, domain2.com ] doesn't take any effect event if it display "Yes" output.

liudf0716, 2017-05-01:
@Tr4cer0ute if reading code of apfree wifidog, u will find pan domains related features are aided by dnsmasq. so u should add dnsmasq-full version to your firmware. noticing must select "build with ipset support"

Tr4cer0ute, 2017-05-01:
I already have it installed 

> Package dnsmasq (2.77test4-1) installed in root is up to date.

The thing is, once I add pad domains, the DHCP server stop working and I can no longer get an IP

liudf0716, 2017-05-02:
@Tr4cer0ute we did not test that dnsmasq version, how about the output of this command:
dnsmasq -C /var/etc/dnsmasq.conf -d

Tr4cer0ute, 2017-05-06:
dnsmasq: failed to create listening socket for port 53: Address in use

liudf0716, 2017-05-07:
u should stop dnsmasq before running dnsmasq in debug mode

closed 2017-06-14 by liudf0716