liudf0716/apfree-wifidog #130: v2.05.1326 is stable?

KazukiShimada, 2017-05-26:
Hi, Liu.

Unfotunately I can't read Chinese. So I can't understand v2.05.1326 description.
https://github.com/liudf0716/apfree_wifidog/releases/tag/2.05.1326
Is v2.05.1326 stable?

regards,

liudf0716, 2017-05-27:
Hi KazukiShimada,

Compared with version 2.03.1061, v2.05.1326 is a pre-release version, after a round of testing, our test team found a few unimportant bugs. If u are demanding, you'd better wait a few days for our official version; if not mind,  u can try this version, and if you find any bug of this version, please let me know by [issues](https://github.com/liudf0716/apfree_wifidog/issues/new). thanks!

regard,

Dengfeng

KazukiShimada, 2017-05-27:
Hi, Liu.

Thank you for your reply.
Ah, I see. I'll wait for new release.

Yes, If I find the issues, I'll report.

Thanks in advance.

closed 2017-05-27 by KazukiShimada