liudf0716/apfree-wifidog #131: ipset need any dependency ?

qianguozheng, 2017-05-27:
add_to_ipset [trustIps] [192.168.3.138] [0]: No such file or directory
2017-05-27 10:22:49.449824|warn|add_to_ipset [trustIps] [192.168.3.139] [0]: Socket operation on non-socket
2017-05-27 10:22:49.450103|warn|add_to_ipset [trustIps] [192.168.3.140] [0]: Socket operation on non-socket
2017-05-27 10:22:49.450383|warn|add_to_ipset [trustIps] [192.168.3.141] [0]: Socket operation on non-socket

liudf0716, 2017-05-27:
the ipset file only provides add&del function, not provides create function.

qianguozheng, 2017-05-27:
Yes. 
Where apfree_wifidog do the create part ?

liudf0716, 2017-05-27:
please read the code.

qianguozheng, 2017-05-27:
ipset_do_command("create " CHAIN_TRUSTED " hash:mac timeout 0 ");
	ipset_do_command("create " CHAIN_ROAM " hash:mac timeout 0 ");
	ipset_do_command("create " CHAIN_UNTRUSTED " hash:mac timeout 0 ");
	ipset_do_command("create " CHAIN_DOMAIN_TRUSTED " hash:ip ");
	ipset_do_command("create " CHAIN_INNER_DOMAIN_TRUSTED " hash:ip ");
	ipset_do_command("create " CHAIN_IPSET_TDOMAIN " hash:ip ");

closed 2017-05-27 by qianguozheng