liudf0716/apfree-wifidog #133: lede17.0.1  iptables1.6出错

vsxen, 2017-05-29:
```
iptables v1.6.1: Couldn't load match `state':No such file or directory

Try `iptables -h' or 'iptables --help' for more information.
[3][Mon May 29 13:52:45 2017][2885](fw_iptables.c:260) iptables command failed(2                                                                                        ): iptables -t filter -A WiFiDog_br-lan_Internet -m state --state INVALID -j DRO                                                                                        P
[3][Mon May 29 13:52:45 2017][2885](https_common.c:17) SSL_CTX_use_certificate_c                                                                                        hain_file failed:

```

rench, 2017-05-29:
1. fork 2.03.1061 to your own rep.
2. replace "-m state --state" to "-m conntrack --ctstate" in src/fw_iptables.c
3. rebuild and package 

我今天搞了一天，终于搞定了。

项目的最版本好像这个已经fix，但是我用最新版编译，程序运行起来直接挂掉，没有任何错误提示。

vsxen, 2017-05-30:
最新版本亲测出错  不是个例

Cherisher, 2017-05-30:
是你自己系统没有编译xt_state.ko模块， 不是bug

vsxen, 2017-05-30:
@Cherisher  是在 kernel/net/netfilter/这里吗？还请赐教

rench, 2017-05-30:
iptables1.6.1 change "-m state --state" to "-m conntrack --ctstate"

Cherisher, 2017-05-31:
```
root@HD:/#iptables -m state --help
iptables v1.6.0
.....省略n行
state match options:
 [!] --state [INVALID|ESTABLISHED|NEW|RELATED|UNTRACKED][,...]
				State(s) to match

```
/lib/modules/4.4.59/xt_state.ko

vsxen, 2017-05-31:
@Cherisher 目前固件内是有这个模块的  我水平不高，还请原谅
```
root@LEDE:~# ls /lib/modules/4.9.20/xt*
/lib/modules/4.9.20/xt_CLASSIFY.ko   /lib/modules/4.9.20/xt_mac.ko
/lib/modules/4.9.20/xt_DSCP.ko       /lib/modules/4.9.20/xt_mark.ko
/lib/modules/4.9.20/xt_HL.ko         /lib/modules/4.9.20/xt_multiport.ko
/lib/modules/4.9.20/xt_LOG.ko        /lib/modules/4.9.20/xt_nat.ko
/lib/modules/4.9.20/xt_NETMAP.ko     /lib/modules/4.9.20/xt_owner.ko
/lib/modules/4.9.20/xt_REDIRECT.ko   /lib/modules/4.9.20/xt_physdev.ko
/lib/modules/4.9.20/xt_TCPMSS.ko     /lib/modules/4.9.20/xt_pkttype.ko
/lib/modules/4.9.20/xt_TPROXY.ko     /lib/modules/4.9.20/xt_quota.ko
/lib/modules/4.9.20/xt_addrtype.ko   /lib/modules/4.9.20/xt_set.ko
/lib/modules/4.9.20/xt_comment.ko    /lib/modules/4.9.20/xt_socket.ko
/lib/modules/4.9.20/xt_conntrack.ko  /lib/modules/4.9.20/xt_state.ko
/lib/modules/4.9.20/xt_dscp.ko       /lib/modules/4.9.20/xt_statistic.ko
/lib/modules/4.9.20/xt_ecn.ko        /lib/modules/4.9.20/xt_tcpmss.ko
/lib/modules/4.9.20/xt_hl.ko         /lib/modules/4.9.20/xt_tcpudp.ko
/lib/modules/4.9.20/xt_length.ko     /lib/modules/4.9.20/xt_time.ko
/lib/modules/4.9.20/xt_limit.ko
root@LEDE:~# wifidog -c  /tmp/wifidog.conf
[6][Wed May 31 04:45:55 2017][2627](conf.c:765) Reading configuration file '/tmp                                                                                        /wifidog.conf'
[6][Wed May 31 04:45:55 2017][2627](gateway.c:768) Forking into background
root@LEDE:~# [6][Wed May 31 04:45:55 2017][2628](gateway.c:502) Setting started_                                                                                        time
[5][Wed May 31 04:45:55 2017][2628](gateway.c:534) Creating web server on 192.16                                                                                        8.123.18:2060
[6][Wed May 31 04:45:55 2017][2628](util.c:92) Closing ICMP socket
[6][Wed May 31 04:45:56 2017][2628](util.c:76) Creating ICMP socket
[6][Wed May 31 04:45:56 2017][2628](firewall.c:238) Initializing Firewall
[6][Wed May 31 04:45:56 2017][2628](wd_util.c:1209) get_ext_iface(): Detected br                                                                                        -lan as the default interface after trying 1
iptables v1.6.1: Couldn't load match `state':No such file or directory

Try `iptables -h' or 'iptables --help' for more information.
[3][Wed May 31 04:45:57 2017][2628](fw_iptables.c:260) iptables command failed(2                                                                                        ): iptables -t filter -A WiFiDog_br-lan_Internet -m state --state INVALID -j DRO                                                                                        P
[3][Wed May 31 04:45:57 2017][2628](https_common.c:17) SSL_CTX_use_certificate_c                                                                                        hain_file failed:

```

Cherisher, 2017-05-31:
You can be sure your modules has be insmod, run as such command:
lsmod |grep xt_state
and use `iptables -m state --help` to see if xt_state.ko is worked

closed 2017-06-01 by vsxen