liudf0716/apfree-wifidog #135: how do i add my own ca,crt,key file????

rench, 2017-05-30:
i sign my own domain with ssl ,but i use copy the file to apfree.ca,apfree.crt,apfree.key .
ssl check success, but ssl_ctx = SSL_CTX_new(SSLv23_method()); print out fail.

there are some debug info;

```
[6][Tue May 30 10:28:44 2017][16638](util.c:92) Closing ICMP socket
[6][Tue May 30 10:28:45 2017][16638](util.c:76) Creating ICMP socket
[6][Tue May 30 10:28:45 2017][16638](firewall.c:238) Initializing Firewall
[6][Tue May 30 10:28:46 2017][16638](https_server.c:295) Updating last_ip IP of server [192.168.2.1] to [192.168.2.1]
[6][Tue May 30 10:28:46 2017][16638](firewall.c:262) Clearing the authservers list
[6][Tue May 30 10:28:46 2017][16638](firewall.c:271) Setting the authservers list
[6][Tue May 30 10:28:46 2017][16638](wd_util.c:121) ONLINE status became ON
[6][Tue May 30 10:28:50 2017][16638](wd_util.c:849) parse domain end, end event_loop [0]
[6][Tue May 30 10:28:50 2017][16638](firewall.c:307) Setting inner trust domains list
[6][Tue May 30 10:28:50 2017][16638](firewall.c:329) Setting user trust domains list
[6][Tue May 30 10:28:50 2017][16638](firewall.c:350) Set trusted maclist
[6][Tue May 30 10:28:50 2017][16638](firewall.c:364) Set untrusted maclist
[3][Tue May 30 10:28:50 2017][16638](simple_http.c:550) SSL_CTX_new failed
[3][Tue May 30 10:28:50 2017][16638](ping_thread.c:100) evhttps_context_init failed, process exit()

```

rench, 2017-05-30:
how do i print the error stack？

liudf0716, 2017-05-31:
@rench u can google openssl's debug api.

rench, 2017-05-31:
@liudf0716  how to use it ? would you show me step?

liudf0716, 2017-05-31:
@rench sorry when  used openssl, I also need google.

closed 2017-06-06 by rench