liudf0716/apfree-wifidog #136: get_cpu_usage failed???

Cherisher, 2017-05-30:
```
static int 
read_cpu_fields (FILE *fp, unsigned long long int *fields)
{
	int retval;
	char buffer[BUF_MAX] = {0};

	if (!fgets (buffer, BUF_MAX, fp)) { 
	 return 0;
	}

	retval = sscanf (buffer, "cpu %Lu %Lu %Lu %Lu %Lu %Lu %Lu %Lu %Lu %Lu", 
							&fields[0], 
							&fields[1], 
							&fields[2], 
							&fields[3], 
							&fields[4], 
							&fields[5], 
							&fields[6], 
							&fields[7], 
							&fields[8], 
							&fields[9]); 
	if (retval < 4) { /* Atleast 4 fields is to be read */
		return 0;
	}
	DebugLog("%s: %s\n", __func__, buf);
	
	DebugLog("retval=%d ", retval);
	for(int i = 0; i < 10; i++){
		DebugLog("%lu ", fields[i]);
	}
	DebugLog("\n");
	return 1;
}
```

结果如下：

[Tue May 30 11:54:31 2017]	 read_cpu_fields: cpu  30050 0 13086 3045700 0 0 14356 0 0 0

[Tue May 30 11:54:31 2017]	 retval=10 
[Tue May 30 11:54:31 2017]	 0 
[Tue May 30 11:54:31 2017]	 0 
[Tue May 30 11:54:31 2017]	 0 
[Tue May 30 11:54:31 2017]	 0 
[Tue May 30 11:54:31 2017]	 0 
[Tue May 30 11:54:31 2017]	 0 
[Tue May 30 11:54:31 2017]	 0 
[Tue May 30 11:54:31 2017]	 0 
[Tue May 30 11:54:31 2017]	 0 
[Tue May 30 11:54:31 2017]	 0 
[Tue May 30 11:54:31 2017]

liudf0716, 2017-05-31:
why u think this result is error?

Cherisher, 2017-05-31:
返回的结果所有fields都为0了， 实际不是， 输出结果很明显了

liudf0716, 2017-05-31:
which file this function read? can u give the full demo?

Cherisher, 2017-05-31:
this is used in function  get_cpu_usage() in src/util.c

liudf0716, 2017-05-31:
I have tested this function, It's ok in apfree wifidog project.

liudf0716, 2017-05-31:
@Cherisher did u try `cat /proc/stat` in your test pc? what's it output.

Cherisher, 2017-05-31:
I have read it and print it：
read_cpu_fields: cpu 30050 0 13086 3045700 0 0 14356 0 0 0

closed 2017-06-13 by Cherisher