liudf0716/apfree-wifidog #139: a android web browser repeat auth

rench, 2017-05-30:
1.Android手机，打开网页的后，进行授权，但是在此访问之前的那个网页，又需要授权。如果在授权后，关闭了浏览器进程，重新打开，访问，就可以正常访问不需要授权了。

2.iPhone 手机，锁屏后，一段时间wifi连接断开。

liudf0716, 2017-05-31:
maybe it's cache problem, someone also encountered this problem before.

rench, 2017-05-31:
@liudf0716  
1.   i visit a http website not https website erverything is ok.
when i visit https website,it will repeate auth.i find my ip in outgoing chain, i don't know why it also redirect wifidog 8443 port, shall we close the ssl socket finally ?

2. i i locking my iphone after few minutes , when i unlock iphone, wifi icon is gone, and re-connecting in few seconds. ???what should cause this problem?

liudf0716, 2017-05-31:
1, it should be cache problem
2, it's iphone's problem

rench, 2017-05-31:
@liudf0716 

2.but if i stop the wifidog ,iphone won't disconnected .

liudf0716, 2017-05-31:
does your iphone pass its auth before lock's screen?

rench, 2017-05-31:
@liudf0716  
yes ,i pass its auth.
and i have other router,iphone won't disconnected too.

liudf0716, 2017-05-31:
@rench if u have pass its auth, I don't think it related with apfree wifidog. I will test and check this problem.

rench, 2017-05-31:
@liudf0716  
thank u，please test and reply to me~

heartache1987, 2017-05-31:
@rench After testing, iPhone will also disconnect WiFi when wifidog is closed

rench, 2017-05-31:
@heartache1987  u mean it's iphone's problem or wifidog iptables chain caused？

liudf0716, 2017-05-31:
heartache1987 has tested following your desciption, but find even stop apfree wifidog, iphone also disconnect WiFi when screen locked

rench, 2017-05-31:
@liudf0716  u mean it's iphone's bug?
but before i  install wifidog ,it's ok.
这就很尴尬了

rench, 2017-06-01:
@liudf0716   @heartache1987  
 i try opkg remove apfree_wifidog , iphone won't diconnect after locking.
you can try it.

liudf0716, 2017-06-01:
@rench there must be some other reason, I can not understand why iphone has passed auth, but has different  performance.

liudf0716, 2017-06-01:
@rench no need to feel embarassed, I believe in code.

heartache1987, 2017-06-01:
@rench You said that a few cases, we have tested here, there are WiFi disconnection phenomenon

rench, 2017-06-01:
@heartache1987  what is your iptables version and linux version?

closed 2017-06-14 by liudf0716