liudf0716/apfree-wifidog #140: can not start ?

xzfth16, 2017-05-31:
root@OpenRouter:~# /usr/bin/wifidog -c /tmp/wifidog.conf
[6][Wed May 31 11:01:00 2017][13228](conf.c:765) Reading configuration file '/tmp/wifidog.conf'
[6][Wed May 31 11:01:00 2017][13228](gateway.c:768) Forking into background
root@OpenRouter:~# [6][Wed May 31 11:01:00 2017][13229](gateway.c:502) Setting started_time
[5][Wed May 31 11:01:00 2017][13229](gateway.c:534) Creating web server on 192.168.10.1:2060
[6][Wed May 31 11:01:00 2017][13229](util.c:92) Closing ICMP socket
[6][Wed May 31 11:01:02 2017][13229](util.c:76) Creating ICMP socket
[6][Wed May 31 11:01:02 2017][13229](firewall.c:238) Initializing Firewall
Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
[3][Wed May 31 11:01:02 2017][13229](fw_iptables.c:260) iptables command failed(4): iptables -t mangle -I PREROUTING 1 -i br-lan -j WiFiDog_br-lan_Outgoing
iptables: Chain already exists.
[3][Wed May 31 11:01:02 2017][13229](fw_iptables.c:260) iptables command failed(1): iptables -t nat -N WiFiDog_br-lan_AuthServers
[3][Wed May 31 11:01:04 2017][13229](https_common.c:17) SSL_CTX_use_PrivateKey_file failed:

root@OpenRouter:~# ps|grep wifidog
13898 root      1484 S    grep wifidog
root@OpenRouter:~#

xzfth16, 2017-05-31:
root@OpenRouter:~# wifidog -c /tmp/wifidog.conf -f -d 7
[6][Wed May 31 11:18:40 2017][20197](conf.c:765) Reading configuration file '/tmp/wifidog.conf'
[7][Wed May 31 11:18:40 2017][20197](conf.c:811) Parsing token: GatewayID, value: 4343434343
[7][Wed May 31 11:18:40 2017][20197](conf.c:811) Parsing token: GatewayInterface, value: br-lan
[7][Wed May 31 11:18:40 2017][20197](conf.c:811) Parsing token: Externalinterface, value: eth0.2
[7][Wed May 31 11:18:40 2017][20197](conf.c:811) Parsing token: AuthServer, value: {
[7][Wed May 31 11:18:40 2017][20197](conf.c:490) Adding 112.74.198.219:80 (SSL: 443) /auth/ to the auth server list
[7][Wed May 31 11:18:40 2017][20197](conf.c:518) Auth server added
[7][Wed May 31 11:18:40 2017][20197](conf.c:811) Parsing token: CheckInterval, value: 60
[7][Wed May 31 11:18:40 2017][20197](conf.c:811) Parsing token: ClientTimeout, value: 72000
[7][Wed May 31 11:18:40 2017][20197](conf.c:811) Parsing token: JsFilter, value: 1
[7][Wed May 31 11:18:40 2017][20197](conf.c:811) Parsing token: WiredPassed, value: 0
[7][Wed May 31 11:18:40 2017][20197](conf.c:811) Parsing token: HTTPDMaxConn, value: 200
[7][Wed May 31 11:18:40 2017][20197](conf.c:811) Parsing token: PoolMode, value: 1
[7][Wed May 31 11:18:40 2017][20197](conf.c:811) Parsing token: ThreadNumber, value: 5
[7][Wed May 31 11:18:40 2017][20197](conf.c:811) Parsing token: QueueSize, value: 20
[7][Wed May 31 11:18:40 2017][20197](conf.c:811) Parsing token: FirewallRuleSet, value: global
[7][Wed May 31 11:18:40 2017][20197](conf.c:553) Adding Firewall Rule Set global
[7][Wed May 31 11:18:40 2017][20197](conf.c:606) Firewall Rule Set global added.
[7][Wed May 31 11:18:40 2017][20197](conf.c:811) Parsing token: FirewallRuleSet, value: validating-users
[7][Wed May 31 11:18:40 2017][20197](conf.c:553) Adding Firewall Rule Set validating-users
[7][Wed May 31 11:18:40 2017][20197](conf.c:589) p1 = [FirewallRule]; p2 = [allow to 0.0.0.0/0]
[7][Wed May 31 11:18:40 2017][20197](conf.c:630) leftover: allow to 0.0.0.0/0
[7][Wed May 31 11:18:40 2017][20197](conf.c:706) Adding Firewall Rule allow (null) port (null) to 0.0.0.0/0
[7][Wed May 31 11:18:40 2017][20197](conf.c:606) Firewall Rule Set validating-users added.
[7][Wed May 31 11:18:40 2017][20197](conf.c:811) Parsing token: FirewallRuleSet, value: known-users
[7][Wed May 31 11:18:40 2017][20197](conf.c:553) Adding Firewall Rule Set known-users
[7][Wed May 31 11:18:40 2017][20197](conf.c:589) p1 = [FirewallRule]; p2 = [allow to 0.0.0.0/0]
[7][Wed May 31 11:18:40 2017][20197](conf.c:630) leftover: allow to 0.0.0.0/0
[7][Wed May 31 11:18:40 2017][20197](conf.c:706) Adding Firewall Rule allow (null) port (null) to 0.0.0.0/0
[7][Wed May 31 11:18:40 2017][20197](conf.c:606) Firewall Rule Set known-users added.
[7][Wed May 31 11:18:40 2017][20197](conf.c:811) Parsing token: FirewallRuleSet, value: auth-is-down
[7][Wed May 31 11:18:40 2017][20197](conf.c:553) Adding Firewall Rule Set auth-is-down
[7][Wed May 31 11:18:40 2017][20197](conf.c:606) Firewall Rule Set auth-is-down added.
[7][Wed May 31 11:18:40 2017][20197](conf.c:811) Parsing token: FirewallRuleSet, value: unknown-users
[7][Wed May 31 11:18:40 2017][20197](conf.c:553) Adding Firewall Rule Set unknown-users
[7][Wed May 31 11:18:40 2017][20197](conf.c:589) p1 = [FirewallRule]; p2 = [allow udp port 53]
[7][Wed May 31 11:18:40 2017][20197](conf.c:630) leftover: allow udp port 53
[7][Wed May 31 11:18:40 2017][20197](conf.c:706) Adding Firewall Rule allow udp port 53 to 0.0.0.0/0
[7][Wed May 31 11:18:40 2017][20197](conf.c:589) p1 = [FirewallRule]; p2 = [allow tcp port 53]
[7][Wed May 31 11:18:40 2017][20197](conf.c:630) leftover: allow tcp port 53
[7][Wed May 31 11:18:40 2017][20197](conf.c:706) Adding Firewall Rule allow tcp port 53 to 0.0.0.0/0
[7][Wed May 31 11:18:40 2017][20197](conf.c:589) p1 = [FirewallRule]; p2 = [allow udp port 67]
[7][Wed May 31 11:18:40 2017][20197](conf.c:630) leftover: allow udp port 67
[7][Wed May 31 11:18:40 2017][20197](conf.c:706) Adding Firewall Rule allow udp port 67 to 0.0.0.0/0
[7][Wed May 31 11:18:40 2017][20197](conf.c:589) p1 = [FirewallRule]; p2 = [allow tcp port 67]
[7][Wed May 31 11:18:40 2017][20197](conf.c:630) leftover: allow tcp port 67
[7][Wed May 31 11:18:40 2017][20197](conf.c:706) Adding Firewall Rule allow tcp port 67 to 0.0.0.0/0
[7][Wed May 31 11:18:40 2017][20197](conf.c:606) Firewall Rule Set unknown-users added.
[7][Wed May 31 11:18:40 2017][20197](conf.c:811) Parsing token: FirewallRuleSet, value: locked-users
[7][Wed May 31 11:18:40 2017][20197](conf.c:553) Adding Firewall Rule Set locked-users
[7][Wed May 31 11:18:40 2017][20197](conf.c:589) p1 = [FirewallRule]; p2 = [block to 0.0.0.0/0]
[7][Wed May 31 11:18:40 2017][20197](conf.c:630) leftover: block to 0.0.0.0/0
[7][Wed May 31 11:18:40 2017][20197](conf.c:706) Adding Firewall Rule block (null) port (null) to 0.0.0.0/0
[7][Wed May 31 11:18:40 2017][20197](conf.c:606) Firewall Rule Set locked-users added.
[7][Wed May 31 11:18:40 2017][20197](conf.c:1626) Parsing string [www.kunteng.org,cloud.kunteng.org,mqtt.kunteng.org,log1.kunteng.org] for trust domains
[7][Wed May 31 11:18:40 2017][20197](conf.c:1632) Locking domain
[7][Wed May 31 11:18:40 2017][20197](conf.c:1632) Domains locked
[7][Wed May 31 11:18:40 2017][20197](conf.c:1651) Adding&Delete [1] trust domain [www.kunteng.org] to&from list
[7][Wed May 31 11:18:40 2017][20197](conf.c:1651) Adding&Delete [1] trust domain [cloud.kunteng.org] to&from list
[... cut ...]
[7][Wed May 31 11:18:44 2017][20197](wdctl_thread.c:169) Creating socket
[7][Wed May 31 11:18:44 2017][20197](wdctl_thread.c:124) Got unix socket 5
[7][Wed May 31 11:18:44 2017][20197](wdctl_thread.c:129) Filling sockaddr_un
[7][Wed May 31 11:18:44 2017][20197](wdctl_thread.c:133) Binding socket (/tmp/wdctl.sock) (15)
[7][Wed May 31 11:18:44 2017][20197](mqtt_thread.c:294) Mosquitto library version : 1.4.8

[3][Wed May 31 11:18:44 2017][20197](https_common.c:17) SSL_CTX_use_PrivateKey_file failed:

liudf0716, 2017-05-31:
it seems x509 cert file error.

xzfth16, 2017-05-31:
How to solve it? I have no idea

liudf0716, 2017-05-31:
use apfree's default cert file and try it.

liudf0716, 2017-05-31:
use apfree's package makefile

xzfth16, 2017-05-31:
Yes i used the default makefile ,  apfree.ca, apfree.crt,apfree.key in the / etc directory

xzfth16, 2017-05-31:
i use openwrt 15.05 use you package openssl and libevent2

liudf0716, 2017-05-31:
which version you use?

xzfth16, 2017-05-31:
Linux version 3.18.36 (xlive@xlive-virtual-machine) (gcc version 4.8.3 (OpenWrt/Linaro GCC 4.8-2014.04 r49403) ) #37 Wed May 31 10:43:53 HKT 2017

liudf0716, 2017-05-31:
I means apfree wifidog's version

xzfth16, 2017-05-31:
PKG_LICENSE:=GPL-2.0
PKG_MAINTAINER:=Dengfeng Liu<liudengfeng@kunteng.org>
PKG_LICENSE_FILES:=COPYING

PKG_SOURCE_PROTO:=git
PKG_SOURCE_URL:=https://github.com/liudf0716/apfree_wifidog.git
PKG_SOURCE_SUBDIR:=$(PKG_NAME)-$(PKG_VERSION)
PKG_SOURCE_VERSION:=$(PKG_VERSION)
PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION)-$(PKG_SOURCE_VERSION).tar.gz
PKG_MIRROR_HASH:=dc1a052bd61c1224dda9548475c1feda9332c40fc2190ca6e4901d432379b51c

xzfth16, 2017-05-31:
PKG_NAME:=apfree_wifidog
PKG_VERSION:=2.03.1061
PKG_RELEASE=1

liudf0716, 2017-05-31:
does /etc/apfree.crt and /etc/apfree.key files exist or not?

xzfth16, 2017-05-31:
yes,Correct presence.

xzfth16, 2017-05-31:
Here is the package that has been installed

opkg list-installed
8021xd - 1
apfree_wifidog - 2.03.1061-1
base-files - 157.2-20161030
blkid - 2.25.2-4
block-mount - 2016-01-10-96415afecef35766332067f4205ef3b2c7561d21
bridge - 1.5-3
busybox - 1.23.2-1
curl - 7.40.0-3.2
ddns-scripts - 2.6.4-1
dnsmasq-full - 2.73-1
dropbear - 2015.67-1
firewall - 2015-07-27
fstools - 2016-01-10-96415afecef35766332067f4205ef3b2c7561d21
hostapd-common - 2015-03-25-1
ip - 4.0.0-1
ip6tables - 1.4.21-1
ipset - 6.24-1
iptables - 1.4.21-1
iptables-mod-conntrack-extra - 1.4.21-1
iptables-mod-extra - 1.4.21-1
iptables-mod-filter - 1.4.21-1
iptables-mod-ipopt - 1.4.21-1
iptables-mod-nat-extra - 1.4.21-1
iptables-mod-tproxy - 1.4.21-1
iw - 4.3-1
iwinfo - 2015-06-01
jshn - 2015-11-08-10429bccd0dc5d204635e110a7a8fae7b80d16cb
jsonfilter - 2014-06-19-cdc760c58077f44fc40adbbe41e1556a67c1b9a9
kernel - 3.18.36-1-d950548af2771c4afff7a249e463dea8
kmod-button-hotplug - 3.18.36-3
kmod-cfg80211 - 3.18.36+2016-01-10-1
kmod-crypto-aead - 3.18.36-1
kmod-crypto-arc4 - 3.18.36-1
kmod-crypto-authenc - 3.18.36-1
kmod-crypto-cbc - 3.18.36-1
kmod-crypto-core - 3.18.36-1
kmod-crypto-deflate - 3.18.36-1
kmod-crypto-des - 3.18.36-1
kmod-crypto-ecb - 3.18.36-1
kmod-crypto-hash - 3.18.36-1
kmod-crypto-hmac - 3.18.36-1
kmod-crypto-iv - 3.18.36-1
kmod-crypto-manager - 3.18.36-1
kmod-crypto-md5 - 3.18.36-1
kmod-crypto-pcompress - 3.18.36-1
kmod-crypto-rng - 3.18.36-1
kmod-crypto-sha1 - 3.18.36-1
kmod-crypto-wq - 3.18.36-1
kmod-fs-ext4 - 3.18.36-1
kmod-fs-vfat - 3.18.36-1
kmod-fuse - 3.18.36-1
kmod-gpio-button-hotplug - 3.18.36-1
kmod-gre - 3.18.36-1
kmod-ifb - 3.18.36-1
kmod-input-core - 3.18.36-1
kmod-ip6tables - 3.18.36-1
kmod-ipsec - 3.18.36-1
kmod-ipt-conntrack - 3.18.36-1
kmod-ipt-conntrack-extra - 3.18.36-1
kmod-ipt-core - 3.18.36-1
kmod-ipt-extra - 3.18.36-1
kmod-ipt-filter - 3.18.36-1
kmod-ipt-ipopt - 3.18.36-1
kmod-ipt-ipset - 3.18.36-1
kmod-ipt-nat - 3.18.36-1
kmod-ipt-nat-extra - 3.18.36-1
kmod-ipt-tproxy - 3.18.36-1
[... 134 lines cut ...]
ppp-mod-pppoe - 2.4.7-6
ppp-mod-pppol2tp - 2.4.7-6
ppp-mod-pptp - 2.4.7-6
pptpd - 1.4.0-2
procd - 2015-10-29.1-d5fddd91b966424bb63e943e789704d52382cc18
resolveip - 2
rpcd - 2016-06-30-23417e94d25570e6d62542bac46edd51e8e0243a
rt2800-usb-firmware - 2016-01-25-52442afee9907bc32a058f22bb3295d040677c26-1
shadowsocks-libev - 3.0.5-1
sqm-scripts - 1.0.7-1
swconfig - 10
tc - 4.0.0-1
ubox - 2015-11-22-c086167a0154745c677f8730a336ea9cf7d71031
ubus - 2015-05-25-f361bfa5fcb2daadf3b160583ce665024f8d108e
ubusd - 2015-05-25-f361bfa5fcb2daadf3b160583ce665024f8d108e
uci - 2015-08-27.1-1
uci2dat - 1
uhttpd - 2015-11-08-fe01ef3f52adae9da38ef47926cd50974af5d6b7
uhttpd-mod-ubus - 2015-11-08-fe01ef3f52adae9da38ef47926cd50974af5d6b7
vsftpd - 3.0.2-4.1
wireless-tools - 29-5
wpa-supplicant-20 - 2.0-0
wpad-mini - 2015-03-25-1
xl2tpd - devel-20150930-1
zlib - 1.2.8-

rench, 2017-05-31:
@liudf0716  your latest makefile package has the wrong apfree.crt, i see the git log ,u delete and commit a new appfree.crt.
i fix this error after using the old apfree.crt

liudf0716, 2017-05-31:
@rench thanks ! I will check it.

liudf0716, 2017-05-31:
@rench I think apfree.crt is the same, I have deleted apfree.crt and apfree.key for mistake, but I think the two certs are same.

xzfth16, 2017-05-31:
The amount, what should I do now? Wait for fix?

liudf0716, 2017-05-31:
@xzfth16 I will try it myself, wait a minute

liudf0716, 2017-05-31:
@rench  is right, the apfree.crt is different, it's my fault. thank you both.

rench, 2017-05-31:
@liudf0716 
There is a very strange question, that is, I use the old version of the apfree.x ssl file can be normal start, and my authorized server is https can also be normal operation. But if I replaced my own signature apfree.x ssl file, wifidog can start normally, but wifidog can not connect to my authorized server https, but can connect http authorization service.

xzfth16, 2017-05-31:
What time can be repaired, this is the first time I use this plugin ，so i want to try the effect.

liudf0716, 2017-05-31:
@xzfth16 we have updated apfree.crt, please try it agian.

liudf0716, 2017-05-31:
@rench  apfree.crt&key were used to redirect https request, it has no relation with auth server.

xzfth16, 2017-05-31:
apfree.crt&key problem has been resolved, I updated the package, the gateway address is not white list by default? After brushing the firmware can not enter the gateway, enter the gateway address, will be redirected, like this:http://entrance.kunteng.org/wifidog/login/?gw_address=192.168.10.1&gw_port=2060&gw_id=000142400891&channel_path=null&ssid=OpenRouter-2.4G-400891&ip=192.168.10.145&mac=00:11:10:21:1c:25&url=http%3A%2F%2F192.168.10.1%2Fcgi-bin%2Fluci

xzfth16, 2017-05-31:
I think ，Is the default gateway address should not be redirected in the whitelist?

liudf0716, 2017-05-31:
@xzfth16  what u mean gateway address, router's address?

xzfth16, 2017-05-31:
Yes，
the router's ip,For example, my router default ip is：192.168.10.1

rench, 2017-05-31:
@liudf0716  i know it has no relation with auth server.but in fact, i  cause the error，and no error msg print out。how cloud i get the ssl error？how to use openssl debug api？

liudf0716, 2017-05-31:
@xzfth16  when u access your router's ip, It's impossible to redirect to auth server.  wifidog only hijack forward flow while accessing router is local in flow.

xzfth16, 2017-05-31:
我想我们都是中国的，英文有时候没准确表达，我意思现在出现开机，浏览器输入路由器的IP会被重定向给认证服务器，之前用的wifidog的时候是没有遇到这种情况的，是不是需要手动设置路由器的IP到免认证白名单？这种情况正常吗?

liudf0716, 2017-05-31:
@xzfth16 sorry to use English, I am not fluent in English either. It's not zhuangbility, cause our issue can help all people, not only Chinese people, when they encoutered the same problem, It also make our supporting more efficient. thanks for your understanding.

liudf0716, 2017-05-31:
@xzfth16 访问路由器的ip是不可能会被劫持的，这个数据流不可能走到被劫持的地方。这个更可能是浏览器的缓存导致。

xzfth16, 2017-05-31:
我将路由器的IP设置到免认证白名单后，路由器地址可以进入了，访问别的网页也弹出认证页面了，但是IP: 192.168.10.145 MAC: 00:11:10:21:1c:25还是无法上网（在状态里面已经看到该设备已经通过认证了），重新开任意的网站都会重新弹出认证页面，wifidog 状态信息：WiFiDog status

Version: 2.03.1061
Uptime: 0d 0h 0m 38s
Has been restarted: no
Internet Connectivity: yes
Auth server reachable: yes
Clients served this session: 2

1 clients connected.

Client 1 status [1]
  IP: 192.168.10.145 MAC: 00:11:10:21:1c:25
  Token: abcdefg
  First Login: 1496217514
  Name: null
  Downloaded: 0
  Uploaded: 0
2 client  1 active .
0 clients unconnected.

xzfth16, 2017-05-31:
# Generated by iptables-save v1.4.21 on Wed May 31 16:06:45 2017
*nat
:PREROUTING ACCEPT [905:276466]
:INPUT ACCEPT [550:35818]
:OUTPUT ACCEPT [494:28686]
:POSTROUTING ACCEPT [247:14042]
:MINIUPNPD - [0:0]
:WiFiDog_br-lan_AuthServers - [0:0]
:WiFiDog_br-lan_Global - [0:0]
:WiFiDog_br-lan_Internet - [0:0]
:WiFiDog_br-lan_Outgoing - [0:0]
:WiFiDog_br-lan_Pass - [0:0]
:WiFiDog_br-lan_Router - [0:0]
:WiFiDog_br-lan_TDomains - [0:0]
:WiFiDog_br-lan_Unknown - [0:0]
:WiFiDog_br-lan_Untrusted - [0:0]
:delegate_postrouting - [0:0]
:delegate_prerouting - [0:0]
:postrouting_lan_rule - [0:0]
:postrouting_rule - [0:0]
:postrouting_wan_rule - [0:0]
:prerouting_lan_rule - [0:0]
:prerouting_rule - [0:0]
:prerouting_wan_rule - [0:0]
:zone_lan_postrouting - [0:0]
:zone_lan_prerouting - [0:0]
:zone_wan_postrouting - [0:0]
:zone_wan_prerouting - [0:0]
-A PREROUTING -j delegate_prerouting
-A PREROUTING -i br-lan -j WiFiDog_br-lan_Untrusted
-A PREROUTING -i br-lan -j WiFiDog_br-lan_Outgoing
-A POSTROUTING -j delegate_postrouting
-A MINIUPNPD -p udp -m udp --dport 52079 -j DNAT --to-destination 192.168.10.145:52079
-A WiFiDog_br-lan_AuthServers -d 112.74.198.219/32 -j ACCEPT
-A WiFiDog_br-lan_Internet -j WiFiDog_br-lan_Pass
-A WiFiDog_br-lan_Internet -m mark --mark 0x2 -j ACCEPT
-A WiFiDog_br-lan_Internet -m mark --mark 0x1 -j ACCEPT
-A WiFiDog_br-lan_Internet -j WiFiDog_br-lan_Unknown
-A WiFiDog_br-lan_Outgoing -j WiFiDog_br-lan_Internet
-A WiFiDog_br-lan_Router -j ACCEPT
-A WiFiDog_br-lan_TDomains -m set --match-set WiFiDog_IPSET_TDomains dst -j ACCEPT
-A WiFiDog_br-lan_TDomains -m set --match-set WiFiDog_br-lan_TDomains dst -j ACCEPT
-A WiFiDog_br-lan_TDomains -m set --match-set WiFiDog_br-lan_ITDomains dst -j ACCEPT
-A WiFiDog_br-lan_Unknown -j WiFiDog_br-lan_AuthServers
-A WiFiDog_br-lan_Unknown -j WiFiDog_br-lan_TDomains
-A WiFiDog_br-lan_Unknown -j WiFiDog_br-lan_Global
-A WiFiDog_br-lan_Unknown -p tcp -m tcp --dport 443 -j REDIRECT --to-ports 8443
-A WiFiDog_br-lan_Unknown -p tcp -m tcp --dport 80 -j REDIRECT --to-ports 2060
-A WiFiDog_br-lan_Untrusted -p tcp -m set --match-set WiFiDog_br-lan_Untrusted src -j REDIRECT --to-ports 44444
-A WiFiDog_br-lan_Untrusted -p udp -m set --match-set WiFiDog_br-lan_Untrusted src -j REDIRECT --to-ports 44444
-A delegate_postrouting -m comment --comment "user chain for postrouting" -j postrouting_rule
-A delegate_postrouting -o br-lan -j zone_lan_postrouting
-A delegate_postrouting -o eth0.2 -j zone_wan_postrouting
-A delegate_prerouting -m comment --comment "user chain for prerouting" -j prerouting_rule
-A delegate_prerouting -i br-lan -j zone_lan_prerouting
-A delegate_prerouting -i eth0.2 -j zone_wan_prerouting
-A zone_lan_postrouting -m comment --comment "user chain for postrouting" -j postrouting_lan_rule
-A zone_lan_prerouting -m comment --comment "user chain for prerouting" -j prerouting_lan_rule
-A zone_wan_postrouting -m comment --comment "user chain for postrouting" -j postrouting_wan_rule
-A zone_wan_postrouting -j MASQUERADE
-A zone_wan_prerouting -j MINIUPNPD
-A zone_wan_prerouting -m comment --comment "user chain for prerouting" -j prerouting_wan_rule
COMMIT
# Completed on Wed May 31 16:06:45 2017
# Generated by iptables-save v1.4.21 on Wed May 31 16:06:45 2017
*raw
:PREROUTING ACCEPT [52054:9914862]
:OUTPUT ACCEPT [36886:13111008]
:delegate_notrack - [0:0]
-A PREROUTING -j delegate_notrack
[... 154 lines cut ...]
-A zone_lan_input -m comment --comment "user chain for input" -j input_lan_rule
-A zone_lan_input -m conntrack --ctstate DNAT -m comment --comment "Accept port redirections" -j ACCEPT
-A zone_lan_input -j zone_lan_src_ACCEPT
-A zone_lan_output -m comment --comment "user chain for output" -j output_lan_rule
-A zone_lan_output -j zone_lan_dest_ACCEPT
-A zone_lan_src_ACCEPT -i br-lan -j ACCEPT
-A zone_wan_dest_ACCEPT -o eth0.2 -j ACCEPT
-A zone_wan_dest_REJECT -o eth0.2 -j reject
-A zone_wan_forward -j MINIUPNPD
-A zone_wan_forward -m comment --comment "user chain for forwarding" -j forwarding_wan_rule
-A zone_wan_forward -p esp -m comment --comment "@rule[3]" -j zone_lan_dest_ACCEPT
-A zone_wan_forward -p udp -m udp --dport 500 -m comment --comment "@rule[4]" -j zone_lan_dest_ACCEPT
-A zone_wan_forward -m conntrack --ctstate DNAT -m comment --comment "Accept port forwards" -j ACCEPT
-A zone_wan_forward -j zone_wan_dest_REJECT
-A zone_wan_input -m comment --comment "user chain for input" -j input_wan_rule
-A zone_wan_input -p udp -m udp --dport 68 -m comment --comment Allow-DHCP-Renew -j ACCEPT
-A zone_wan_input -p icmp -m icmp --icmp-type 8 -m comment --comment Allow-Ping -j ACCEPT
-A zone_wan_input -p igmp -m comment --comment Allow-IGMP -j ACCEPT
-A zone_wan_input -m conntrack --ctstate DNAT -m comment --comment "Accept port redirections" -j ACCEPT
-A zone_wan_input -j zone_wan_src_ACCEPT
-A zone_wan_output -m comment --comment "user chain for output" -j output_wan_rule
-A zone_wan_output -j zone_wan_dest_ACCEPT
-A zone_wan_src_ACCEPT -i eth0.2 -j ACCEPT
COMMIT
# Completed on Wed May 31 16:06:45 2017

xzfth16, 2017-05-31:
在iptables中也看到wifidog添加了192.168.10.145设备，但是开任意网址还是会被重定向。

liudf0716, 2017-05-31:
if u router's ip is 192.168.10.1, it's impossible when u access 192.168.10.1, it was hijjacked to login page.

rench, 2017-05-31:
a haha,it maybe the cache problem???????

closed 2017-06-02 by xzfth16