liudf0716/apfree-wifidog #144: Temporary Whitelist

UnKnoWn-Consortium, 2017-06-12:
Hi!

I have read in this issue #34 that "when it receive second captive.apple.com visit for more than 20 seconds, it will automatically add this device into temporary white mac list for a minute". 

I want to allow two to three minutes for guests to finish the proper oAuth process. So I would like to know if there is any way to access this temporary white mac list or modify this behavior. If it is hard-coded, could you please point me to the relevant parts in the source? 

Thank you very much.

liudf0716, 2017-06-13:
@UnKnoWn-Consortium 
In fact, u can this web api :
 ‘http://{{gw_address}}:{{gw_port}}/wifidog/temporary_pass?mac={{mac}}’
it can add the mac's device to whitelist for a minute

UnKnoWn-Consortium, 2017-06-13:
That's great! Thanks a lot!

closed 2017-06-13 by UnKnoWn-Consortium