liudf0716/apfree-wifidog #150: allow 443 port

Tr4cer0ute, 2017-06-16:
How can I allow 443 port? which option should I use?

JJEU, 2017-06-16:
in original wifidog, it's in wifidog.conf
search for this section : FirewallRuleSet global {
add this :
FirewallRule allow udp port 443

for appfree it may be different, or maybe it's the same. just test it and let us know

Tr4cer0ute, 2017-06-16:
Thanks but if its the same I would not post it here ;-) anyway I will re
check

JJEU, 2017-06-18:
let me know if you find out because i would have the same question

zainbouziane, 2017-06-19:
I can confirm that this does not work. The wifidog.conf is in /tmp and will be overwritten during start of wifidog.
You should set it in /etc/config/wifidog but i dont know the option i have to use to allow it

liudf0716, 2017-06-20:
@zainbouziane 
u are right.  when apfree wifidog start, it will overwritten /tmp/wifidog.conf. 
@Tr4cer0ute , @JJEU  if  modified /tmp/wifidog.conf, u can use `wifidog -c /tmp/wifidog.conf -d 0` to verify whether your rule is valid or not instead of using `/etc/init.d/wifidog start`. 
we will add this feature to '/etc/config/wifidog' and  '/etc/init.d/wifidog' as soon as possible.
thank u all!

caizhibang, 2017-06-20:
@Tr4cer0ute @JJEU in lastest apfree version ,you can add " list firewall_rule_unknown_users 'allow tcp port 443' " in /etc/config/wifidog  to allow port 443 . And then /etc/init.d/wifidog restart to let it enable.

Tr4cer0ute, 2017-06-20:
@caizhibang I will give it a try.
Thanks

closed 2017-06-27 by liudf0716