liudf0716/apfree-wifidog #151: Compile error

zainbouziane, 2017-06-19:
Hi,

While compiling i get this error:
`/home/zain/lede/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/lib/gcc/mips-openwrt-linux-musl/5.4.0/../../../../mips-openwrt-linux-musl/bin/ld: copy reloc against protected `__stack_chk_guard' is dangerous
CMakeFiles/wifidog.dir/simple_http.c.o: In function `evhttps_request':
simple_http.c:(.text+0x9e2): undefined reference to `bufferevent_openssl_set_allow_dirty_shutdown'
simple_http.c:(.text+0x9f2): undefined reference to `evhttp_connection_base_bufferevent_new'
CMakeFiles/wifidog.dir/https_server.c.o: In function `thread_https_server':
https_server.c:(.text+0x5b8): undefined reference to `evhttp_set_bevcb'
collect2: error: ld returned 1 exit status
src/CMakeFiles/wifidog.dir/build.make:797: recipe for target 'src/wifidog' failed
make[6]: *** [src/wifidog] Error 1
make[6]: Leaving directory '/home/zain/lede/build_dir/target-mips_24kc_musl/apfree_wifidog/apfree_wifidog-2.03.1061'
CMakeFiles/Makefile2:124: recipe for target 'src/CMakeFiles/wifidog.dir/all' failed
make[5]: *** [src/CMakeFiles/wifidog.dir/all] Error 2
make[5]: Leaving directory '/home/zain/lede/build_dir/target-mips_24kc_musl/apfree_wifidog/apfree_wifidog-2.03.1061'
Makefile:129: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/zain/lede/build_dir/target-mips_24kc_musl/apfree_wifidog/apfree_wifidog-2.03.1061'
Makefile:66: recipe for target '/home/zain/lede/build_dir/target-mips_24kc_musl/apfree_wifidog/apfree_wifidog-2.03.1061/.built' failed
make[3]: *** [/home/zain/lede/build_dir/target-mips_24kc_musl/apfree_wifidog/apfree_wifidog-2.03.1061/.built] Error 2
make[3]: Leaving directory '/home/zain/lede/package/apfree_wifidog'
package/Makefile:109: recipe for target 'package/apfree_wifidog/compile' failed
make[2]: *** [package/apfree_wifidog/compile] Error 2
make[2]: Leaving directory '/home/zain/lede'
package/Makefile:105: recipe for target '/home/zain/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/zain/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zain/lede'
/home/zain/lede/include/toplevel.mk:207: recipe for target 'world' failed
make: *** [world] Error 2
`

Hope you can help.

Greets!

liudf0716, 2017-06-19:
u should use libevent version 2.1.8

zainbouziane, 2017-06-19:
Thanks! It works now

closed 2017-06-19 by zainbouziane