liudf0716/apfree-wifidog #152: Subdomain wildcards not working

zainbouziane, 2017-06-19:
Hi,

I want to whitelist youtube.com for example. But youtube uses many subdomains for streaming its video.
So i tried *.youtube.com but it doesnt work.
Is there a wildcard character for subdomains?

Greets

liudf0716, 2017-06-20:
@zainbouziane 
I think u can try pan domains feature instead of wildcard subdomains. which apfree wifidog does not support. If u want whitelist youtube.com, u can use the following command:
`wdctl add_trusted_pdomains youtube.com`
PS: pan domains feature depends on dnsmasq's ipset support, so your openwrt&lede should select dnsmasq-full with IPset support

as for MQTT, apfree wifidog use mqtt to remotely add&del&show apfree wifidog's related rule, such as add&clear&show  trusted domains|pdomains|mac and etc. here is details
https://github.com/liudf0716/apfree_wifidog/blob/master/apfree_wifidog_mqtt_protocol

zainbouziane, 2017-06-20:
I installed dnsmasq-full and ipset was already installed.
I executed `wdctl add_trusted_pdomains youtube.com` and after that `wdctl restart`
But when i visit youtube.com or his subdomains i am redirected to the login page.
How can i fix this?

liudf0716, 2017-06-21:
@zainbouziane  
firstly, after executing wdctl add_trusted_pdomains youtube.com, no need to restart wifidog. 
secondly, for my fault, `wdctl restart` does not work properly, it has bug, pls don't use it.

liudf0716, 2017-06-21:
u can try `ipset list` to debug this problem.

zainbouziane, 2017-06-26:
Thanks, it works

zainbouziane, 2017-06-27:
Yet, it still doesnt work properly. Sometimes if i boot the router i still cant access facebook etc.
Here is the result of `ipset list`:

```
Name: WiFiDog_br-lan_Trusted
Type: hash:mac
Revision: 0
Header: hashsize 1024 maxelem 65536 timeout 0
Size in memory: 64
References: 1
Members:

Name: WiFiDog_br-lan_Roam
Type: hash:mac
Revision: 0
Header: hashsize 1024 maxelem 65536 timeout 0
Size in memory: 64
References: 1
Members:

Name: WiFiDog_br-lan_Untrusted
Type: hash:mac
Revision: 0
Header: hashsize 1024 maxelem 65536 timeout 0
Size in memory: 64
References: 2
Members:

Name: WiFiDog_br-lan_TDomains
Type: hash:ip
Revision: 4
Header: family inet hashsize 1024 maxelem 65536
Size in memory: 64
References: 2
Members:

Name: WiFiDog_br-lan_ITDomains
Type: hash:ip
Revision: 4
Header: family inet hashsize 1024 maxelem 65536
Size in memory: 224
References: 2
Members:
114.112.99.138
114.112.99.144
114.112.99.249
114.112.99.142

Name: WiFiDog_IPSET_TDomains
Type: hash:ip
Revision: 4
Header: family inet hashsize 1024 maxelem 65536
Size in memory: 64
References: 2
Members:
```
`
/etc/config/wifidog`:

```

config wifidog
	option gateway_interface 'br-lan'
	option auth_server_hostname '***'
	option auth_server_port '80'
	option auth_server_path '/'
	option check_interval '60'
	option client_timeout '72000'
	option httpd_max_conn '200'
	option pool_mode '1'
	option thread_number '5'
	option queue_size '20'
	option wired_passed '0'
	option trusted_domains 'fonts.gstatic.com'
	option trusted_pan_domains 'googlevideo.com,facebook.net,facebook.com,youtube.com'
```

liudf0716, 2017-06-28:
@zainbouziane  what's the result of `ll /tmp/dnsmasq.d/` and cat the files in that directory

closed 2017-06-29 by zainbouziane