liudf0716/apfree-wifidog #153: make package/apfree_wifidog/compile V=s error with openwrt 15

mrsantran, 2017-06-24:
[image]

When i build apfree_wifidog on centos with openwrt 15.
Have error.
How to resolved ?

liudf0716, 2017-06-24:
Need more information

KazukiShimada, 2017-06-25:
You failed to build toolchain. You didn't even reach wifidog compile yet.
Try lede buildroot.
https://github.com/lede-project/source

mrsantran, 2017-06-25:
I'm do all step in guide for  openwrt  and lede.

>https://github.com/liudf0716/apfree_wifidog/wiki/%E5%9F%BA%E4%BA%8ELEDE%E7%BC%96%E8%AF%91Apfree_wifidog

> https://github.com/liudf0716/apfree_wifidog/issues/86

try:
make package/apfree_wifidog/compile V=s
make package/apfree_wifidog/compile -j1 V=s
make package/apfree_wifidog/compile -j2 V=s
make package/apfree_wifidog/compile -j3 V=s
make package/apfree_wifidog/compile -j4 V=s
make V=s
...

but error at all.
video:
https://www.facebook.com/santran.686/videos/242151116271660/

liudf0716, 2017-06-26:
KazukiShimada are right, before compile apfree wifidog, u should have toolchain built, in order to do that, u firstly should `make V=s` before compile apfree wifidog.

mrsantran, 2017-06-26:
Thanks.

But run make V=s same error toolchain

liudf0716, 2017-06-27:
that's LEDE&OPENWRT's problem, u can read their document to solve it.

lcyenter, 2017-06-29:
Linking C executable wifidog
CMakeFiles/wifidog.dir/wd_util.c.o: In function `wd_gethostbyname':
wd_util.c:(.text+0x2804): warning: gethostbyname is obsolescent, use getnameinfo() instead.
/home/lcy/openwrt/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/usr/lib/libuci.so: warning: the use of `mktemp' is dangerous, better use `mkstemp'
CMakeFiles/wifidog.dir/simple_http.c.o: In function `evhttps_request':
simple_http.c:(.text+0x10c4): undefined reference to `bufferevent_openssl_set_allow_dirty_shutdown'
simple_http.c:(.text+0x10e0): undefined reference to `evhttp_connection_base_bufferevent_new'
CMakeFiles/wifidog.dir/https_server.c.o: In function `thread_https_server':
https_server.c:(.text+0x9bc): undefined reference to `evhttp_set_bevcb'
collect2: error: ld returned 1 exit status
make[5]: *** [src/wifidog] Error 1
make[5]: Leaving directory `/home/lcy/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/apfree_wifidog/apfree_wifidog-2.03.1061'
make[4]: *** [src/CMakeFiles/wifidog.dir/all] Error 2
make[4]: Leaving directory `/home/lcy/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/apfree_wifidog/apfree_wifidog-2.03.1061'
make[3]: *** [all] Error 2
make[3]: Leaving directory `/home/lcy/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/apfree_wifidog/apfree_wifidog-2.03.1061'
make[2]: *** [/home/lcy/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/apfree_wifidog/apfree_wifidog-2.03.1061/.built] Error 2
make[2]: Leaving directory `/home/lcy/openwrt/package/apfree_wifidog'
make[1]: *** [package/apfree_wifidog/compile] Error 2
make[1]: Leaving directory `/home/lcy/openwrt'
make: *** [package/apfree_wifidog/compile] Error 2

I searched this problem a long time but can't find any solutions,how to solve this problem??

liudf0716, 2017-06-29:
@lcyenter read this issue
https://github.com/liudf0716/apfree_wifidog/issues/138

KazukiShimada, 2017-06-29:
Hi.
First of all, you should read README on top page.
https://github.com/liudf0716/apfree_wifidog

lcyenter, 2017-06-29:
My god..I finally knew it,I don't need the https,I think it's not necessary,so it causes me failed many times..Thanks a lot!!!

closed 2017-07-12 by liudf0716