liudf0716/apfree-wifidog #155: Error: We did not get a valid answer from the central server

zainbouziane, 2017-06-26:
Hi,

Wifidog returns a page with the error: `Error: We did not get a valid answer from the central server`
While my auth server correctly returns: `Auth: 1` as i see in the webserver logs

What can be the problem?

liudf0716, 2017-06-27:
please read this article
https://github.com/liudf0716/apfree_wifidog/wiki/apfree-wifidog%E8%AE%A4%E8%AF%81%E8%BF%94%E5%9B%9E-We-did-not-get-a-valid-answer-from-the-central-server

zainbouziane, 2017-06-27:
My /auth file contains this: 
`<?PHP
echo "Auth: 1";`

But i still get the same error `Error: We did not get a valid answer from the central server`

liudf0716, 2017-06-28:
u can debug apfree wifidog like this
wifidog -c /tmp/wifidog.conf -f -d 7

zainbouziane, 2017-06-28:
Thanks, found the error. My server was returning a 301 error due to a rule in my .htaccess

closed 2017-06-28 by zainbouziane