liudf0716/apfree-wifidog #161: Can't not Capture Https.

LeonKendy, 2017-07-07:
Hi,
When i access https by mobile phone running os android.I don't see capture HTTPS URL request.Example : google, and facebook.....
[image]

liudf0716, 2017-07-07:
@LeonKendy In fact, from your pic, apfree has captured https request.

LeonKendy, 2017-07-07:
@liudf0716 
I mean it's not redirect to page login to access internet .

liudf0716, 2017-07-07:
@LeonKendy  I know what u mean, but https can not hijack like http, that why it's https. https client must notify user when it found  web's cert is illegal, of course, apfree use illegal cert for https client.

LeonKendy, 2017-07-07:
hi @liudf0716 ,
thank you for your answer and i have one more question.
How to show Capture when case user accept https for them login active internet.
Because am my laptop when accept https it will show one more tab login for user login but my phone it not.

closed 2017-07-08 by LeonKendy

rahacker2006, 2017-10-18:
@LeonKendy you resolve the issue? please confirm

rahacker2006, 2017-10-18:
And how did you do it? @LeonKendy