liudf0716/apfree-wifidog #164: Update Apfree_wifidog-v2.06.1395 start but not run.

LeonKendy, 2017-07-08:
I installed apfree_wifidog ver latest releases complete. Then I run the command /etc/init.d/wifidog start and ps | Grep wifidog. And I detected the wifidog process not active. I checked the system logs apart from the web interface, the following message was detected.
I removed the installation of Apfree_wifidog ver latest releases and reinstalled Apfree_wifidog ver 2.03.1061. Then run the command /etc/init.d/wifidog start and ps | Grep wifidog sees active wifidog process
Look forward to somebody's help :(
[image]
[image]
[image]

LeonKendy, 2017-07-08:
I check command /usr/bin/wifidog -c /tmp/wifidog.conf -f -d 0 and i recived message faild Ip Tables .
[image]
[image]

LeonKendy, 2017-07-08:
I check command /usr/bin/wifidog -c /tmp/wifidog.conf -f -d 10 and i recived message ip tables command faild 
[image]
[image]

liudf0716, 2017-07-12:
when wifidog start, it will initialize iptables rules : flush and delete possible wifidog rule, if it firstly start, wifidog rule does not exist, so u can see many info about iptables execute failed. it's not problem.

closed 2017-07-16 by LeonKendy