liudf0716/apfree-wifidog #165: Keep browser opened after connection is allowed

iffifan, 2017-07-17:
Hello,

We are facing this issue. When a user connects and is authenticated by auth server, browser window is closed automatically on android 5+. It seems like it opens captive portal in some special browser just t get user authenticated and closes that browser immediately after authentication is done. We want to keep it open so that we can show some advertisement after user is authenticated. We want user to click on advertisement and browse pages.

Any idea?

liudf0716, 2017-07-18:
I have never encountered such phenomenon, what's your phone brand?

JJEU, 2017-07-18:
it's a completely normal behavior.

My advice is to put the adds directly on the authentication page and not after. You can build your auth page with a CMS so you could personalise the ads at each venue

Reason : 
The new OSes are able to detect captive portals. when they do so, they open a temporary browser. when they detect that the internet is accessible, they close it automatically

You can fool them by etheir : 
- allowing the adresses they probe. this way the phone won't be able to detect that there is a captive portal. Drawback : the user would have to open manually it's browser, AND navigate manually to an http (not https) web page, to be redirected to your auth page

- blocking the adresses they probe : this way the phone will detect the portal page and open the temporary browser.
After the authentication, the adresses will remain blocked and the phone won't be able to know that the internet is accessible. so the page will remain open.
drawback : there could be strange behavior of the phones. for exemple after some time they could revert to 3G/4G if they are unable to detect the internet with the wifi.
also some adresses are really generic and could block other services from google or apple.

you can find more information about this problem on google. also do a lot of research in order to get all the possible adresses.
https://www.google.ca/search?q=wifi+captive+portal+page+close+automatically

https://stackoverflow.com/questions/34933146/prevent-captive-portal-auto-close-after-authentication-android

https://stackoverflow.com/questions/26063547/wi-fi-captive-portals-successfully-authenticated-page-is-closed-automatically-i

this is also a way to hack the phones with this behavior
https://arstechnica.com/security/2015/06/evil-wifi-captive-portal-could-fool-users-into-giving-up-apple-pay-data/

closed 2017-08-24 by liudf0716