liudf0716/apfree-wifidog #166: Authentication server check bug.

xzfth16, 2017-07-31:
When I build the authentication server on the X86 router(Meaning the router itself is the authentication server), I will not be able to detect server bugs
error log:
[6][Mon Jul 31 21:38:49 2017][30953](centralserver.c:412) Sorry, internet is not available!
[3][Mon Jul 31 21:38:49 2017][30953](centralserver.c:329) Failed to connect to any of the auth servers
http error page display:
"Error!", "Error: We did not get a valid answer from the central server"

PS:Original wifidog no such error.

liudf0716, 2017-08-01:
when apfree wifidog connect auth server, it will detect whether the device is online or not, if not, it will not connect auth server.

xzfth16, 2017-08-01:
if auth server = itself ， can not detect itself ！wifidog no such problem，apfree_wifidog has this problem。

xzfth16, 2017-08-01:
You can try it yourself，I have tested many times。

liudf0716, 2017-08-01:
your device is online?

xzfth16, 2017-08-01:
Suer!Correct access to the Internet.

closed 2017-12-04 by liudf0716