liudf0716/apfree-wifidog #167: missing file mosquitto.h

etmad8, 2017-08-14:
/build_dir/target-mips_24kc_musl/apfree_wifidog/apfree_wifidog-2.03.1061/src/mqtt_thread.c:36:23: fatal error: mosquitto.h: No such file or directory

missing file

liudf0716, 2017-08-17:
please compile mosquitto firstly

serezhka, 2017-08-29:
Same here.

@liudf0716 , could you please explain where and how mosquitto.h have to be placed? 
libmosquitto-dev & mosquitto-dev are installed

-- 
There are other build errors, for example 
http://downloads.openwrt.org/sources/linux-firmware-2016-01-25-52442afee9907bc32a058f22bb3295d040677c26.tar.xz isn't available .. 
It can be downloaded from http://dev.gateworks.com/sources/
and placed in [openwrt]/dl/

liudf0716, 2017-08-30:
@serezhka u need to compile mosquitto, not just install it
make package/mosquitto/compile V=s

zcfrank1st, 2017-08-31:
After `make package/mosquitto/compile V=s`, compiling apfree_wifidog is still error :
Package apfree_wifidog is missing dependencies for the following libraries:
libmosquitto.so.1...

liudf0716, 2017-08-31:
@zcfrank1st  do u choose libmosquitto?
make menuconfig ---> choose libmosquitto

zcfrank1st, 2017-09-01:
done,  thanks

closed 2017-09-01 by liudf0716

stevetao74, 2019-03-28:
@nuwanarti it's suiting to apt install, good ¡