liudf0716/apfree-wifidog #168: `sh: /usr/bin/ktpriv: not found` error

zcfrank1st, 2017-09-01:
When running `wifidog -f -c /tmp/wifidog.conf -d 4` with an auth server, it is ok for a while and then the terminal outputs `sh: /usr/bin/ktpriv: not found`...  And the page can not vist, it jumps to the login page again

liudf0716, 2017-09-01:
please use latest release version. ktpriv is removed.

zcfrank1st, 2017-09-01:
I already have used the latest version to compile...

liudf0716, 2017-09-01:
no need to care this problem. I will comment it soon.

zcfrank1st, 2017-09-01:
ok，I will wait, thank you

zcfrank1st, 2017-09-01:
你的master分支上的package目录里wifidog的makefile软件版本没有改，我用的还不是最新的，我改下用最新的再试试。。。

zcfrank1st, 2017-09-01:
不行，编译报错，还是等你发版本吧

liudf0716, 2017-09-01:
can u provide compile error info?

zcfrank1st, 2017-09-01:
../arm-openwrt-linux-uclibcgnueabi/bin/ld: cannot find -liptext
../arm-openwrt-linux-uclibcgnueabi/bin/ld: cannot find -liptext4

zcfrank1st, 2017-09-01:
刚刚看了一下你master上的代码，貌似废弃的判断是有线无线的方法还是有地方调用。。。

liudf0716, 2017-09-02:
ktpriv mainly used to decide whether client is wireless or not.

zcfrank1st, 2017-09-02:
我又重新编译试了一下，用最新的版本（v2.06.1395 以及 openwrt Chaos Calmer 15.05.1）还是报错这个问题，然后又要再通过authserver登录验证一遍。。。

zcfrank1st, 2017-09-02:
希望能尽快修复，谢谢~  记得把 apfree_wifidog 的 makefile 版本改下，要不然还是 `PKG_VERSION:=2.03.1061`

liudf0716, 2017-09-06:
@zcfrank1st  fixed it in master

zcfrank1st, 2017-09-07:
Thanks, I will try ~

zcfrank1st, 2017-09-07:
Also have some problems...  the logs below :
[5][Thu Sep  7 09:09:56 2017][3189](firewall.c:510) 192.168.33.118 - Denied. Removing client and firewall rules
when I login successfully, about one minute later, I was kicked away by the iptables...

liudf0716, 2017-09-07:
why u think it's a problem?

zcfrank1st, 2017-09-08:
because I set 10 hours expired，but it expires just about one minute... so i think it is a problem.
but i work out that there was something wrong with my own auth server, i fix the auth server return code,then it works 

thanks for your patience~

by the way ,can you update the ap-wifidog make file? it is still the old version ...

liudf0716, 2017-09-08:
@zcfrank1st In fact, u can update it yourself by providing PR:
https://github.com/liudf0716/apfree_wifidog/blob/master/CONTRIBUTING.md

closed 2017-12-04 by liudf0716