liudf0716/apfree-wifidog #169: Wifidog configuration issue

AvalonTechnologies, 2017-09-06:
When I try to run the wifidog command:
```
[6][Wed Sep  6 08:14:06 2017][1657](conf.c:765) Reading configuration file '/etc/wifidog.conf'
[3][Wed Sep  6 08:14:06 2017][1657](conf.c:768) Could not open configuration file '/etc/wifidog.conf', exiting...
```
When I try to load configuration file:
```
[6][Wed Sep  6 08:29:05 2017][1711](conf.c:765) Reading configuration file 'wifidog.conf'
[3][Wed Sep  6 08:29:05 2017][1711](conf.c:348) wifidog.conf: line 1: Bad configuration option: config
[3][Wed Sep  6 08:29:05 2017][1711](conf.c:974) Bad option on line 1 in wifidog.conf.
[3][Wed Sep  6 08:29:05 2017][1711](conf.c:975) Exiting...
```
This is what I get when I do ps | grep wifidog
` 1715 root      1036 R    grep wifidog`

I followed the steps to compile wifidog, had to update a few packages and install fping manually. But it does not load config file so have I screwed up somewhere in compiling wifidog?

liudf0716, 2017-09-07:
@AvalonTechnologies please use our wifidog package for openwrt&LEDE
when wifidog startup, it will read its config file, our wifidog.init file will generate wifidog.conf  to /tmp directory when started wifidog.

AvalonTechnologies, 2017-09-07:
Ok ive tried many things so far and wifidog is not working with LEDE. Here is what im using:
LEDE version 17.01.2
Linksys WRT1200ac
latest version of apfree_wifidog
I got this configuration working once by manually putting this file in https://github.com/liudf0716/apfree_wifidog/blob/master/wifidog.conf and starting the service from the command `/etc/init.d/wifidog start`
The router successfully redirects me to the auth server/portal and lets me authenticate. But the problem with this is that it does not intercept HTTPS connections which apfree_wifidog is meant to be able to do. 
Another thing with this is when i do ps | grep wifidog it does not show any running service so I think its not actually running.
Also when I restart the router wifidog does not start.

Ok onto your suggestion and following https://github.com/liudf0716/apfree_wifidog#getting-started, this guide doesnt provide much information sadly so im not sure what to do but i certainly dont get this result when I run it after i do opkg install apfree_wifidog_2.03.1061-1_arm_cortex-a9_vfpv3.ipk
```
root@lede:~# ps | grep wifidog
 1406 root      6532 S    /usr/bin/wifidog -c /tmp/wifidog.conf -f -d 0
```
Another thing I had to do was install fping_4.0-2_arm_cortex-a9_vfpv3.ipk before I could get wifidog to run.

Is there some pre-requisites that must be installed before I can install apfree-wifidog on LEDE? I think the guide is missing some steps on installation.
This is all the files I get after I compile http://rgho.st/8BxZZvDT9
Can you give me some detailed steps that im missing perhaps to complete my installation.

liudf0716, 2017-09-08:
@AvalonTechnologies  I think u have compiled apfree wifidog successly according to the following guid:
https://github.com/liudf0716/apfree_wifidog#getting-started
since u have found wifidog process through 'ps|grep wifidog', I think apfree wifidog should run properly without problem.  
In our apfree wifidog Makefile, u can find all depends package in its DEPENDS. in order to support https redirect, we need libevent-2.1.8, since LEDE and openwrt did not accept my PR, when u compile apfree wifidog, u should replace original libevent package with our libevent package,  which provided in our package directory.
someone has reported to me, when compiled apfree wifidog in latest LEDE version, they  also encountered the problem that  apfree wifidog could not start. after running wifidog -c /tmp/wifidog.conf -f -d 7 , it seems related to iptable's library version, so I suggest u try it in openwrt cc.

closed 2017-12-04 by liudf0716

cuongbn, 2017-12-17:
@AvalonTechnologies 
Are you edit file /etc/config/wifidog  variable enable 0 => 1.