liudf0716/apfree-wifidog #170: Wifidog not active version Latest because Fw3_iptc.c

LeonKendy, 2017-09-08:
Wifidog works normal old version. But the trust mac option does not work as expected. I see a new update fix this problem and download the build again.
After reinstalling the latest version, the wifi does not work anymore.
I run cli: / usr / bin / wifidog -c /tmp/wifidog.conf -d 4 then I get the following error message.
"(fw3_iptc.c: 646) iptc_create_chain (): No chain / target / match by that name

 (fw3_iptc.c: 646) iptc_create_chain (): No chain / target / match by that name

(fw3_iptc.c: 646) iptc_create_chain (): No chain / target / match by that name

(fw3_iptc.c: 646) iptc_create_chain (): No chain / target / match by that name

(fw3_iptc.c: 697) iptc_delete_chain (): No chain / target / match by that name

(fw3_iptc.c: 697) iptc_delete_chain (): No chain / target / match by that name

(fw3_iptc.c: 697) iptc_delete_chain (): No chain / target / match by that name

 (fw3_iptc.c: 697) iptc_delete_chain (): No chain / target / match by that name

(fw3_iptc.c: 697) iptc_delete_chain (): No chain / target / match by that name "

@liudf0716 .Can you help me fix this problem. Tks 
[image]

liudf0716, 2017-09-12:
@LeonKendy thanks for your report
can u provide me the information about your platform and its version, for example lede 17.02 or openwrt cc

liudf0716, 2017-09-12:
@LeonKendy what u provide information can not prove apfree wifidog has bug.
when apfree wifidog started, it will clear all iptable or ipset rule; if  wifidog started first time, there was no rules, when u flush and delete a non-exist rule, it will report above information.

gigibox, 2017-09-12:
Hi @LeonKendy 
    This is normal and these messages can be ignored. firewall rule is cleared regardless of errors  while you start/restart the wifidog. 
    fw3_iptc.c: 646 debug information is a bug, it should be "iptc_flush_chain".  thanks a lot!

   about trust mac option, you can check dnsmasq version, `dnsmasq -v`, need ipset support.

LeonKendy, 2017-09-12:
@liudf0716  @gigibox  Tks for your answer.
Firstly, I user LEDE Reboot SNAPSHOT r4610-97eb8ab new than 17.01.2.
Secondly, If this problem is normal and these messages can be ignored, my wifidog process is working normally. But I run the command: ps | grep wifidog and do not see anything.
I try to rebuild the file you just updated try to see the results.
Thanks again for the above explanation that really helped me understand the problem.

LeonKendy, 2017-09-12:
wifidog start/stop and restart still not working 
[image]

liudf0716, 2017-09-12:
@LeonKendy  pls use wifidog -c /tmp/wifidog.conf -f -d 4

LeonKendy, 2017-09-12:
@liudf0716  of course . I run wifidog -c /tmp/wifidog.conf -f -d 4
[image]

closed 2017-12-04 by liudf0716