liudf0716/apfree-wifidog #171: How to open firewall with ip and port

tidus30691nexus5, 2017-09-12:
Please help me how to open firewall with ip and port 
Exmaple: am want open ip 192.168.1.1 with port 443

please help me. thank you

liudf0716, 2017-09-12:
we have trusted ip and mac, but it seems we did not provide trusted ip&port feature. I wonder why u need this feature?

tidus30691nexus5, 2017-09-12:
Hi @liudf0716 
Because am read some document should open range IP + port 443 to have captive on IOS . My iphone 5 not have captive open but all android have. So am am think need open range ip + port ...

liudf0716, 2017-09-12:
apfree wifidog has treated ios captive, u didn't need to do more.

tidus30691nexus5, 2017-09-12:
but why my iphone not have open captive ( my version is 10.3.2 iphone5). Only load auth page when access  
website not https. 
Please help me solve it.
thank you

closed 2017-12-04 by liudf0716