liudf0716/apfree-wifidog #172: failed to start

KazukiShimada, 2017-09-14:
Hi.
I tried apfree on latest lede with x86 geode board. Apfree failed to start with segfault.

root@LEDE:/etc/init.d# ps|grep wifidog
 4442 root       972 S    grep wifidog
root@LEDE:/etc/init.d# /etc/init.d/wifidog start
root@LEDE:/etc/init.d# ps |grep wifidog
 4629 root       972 S    grep wifidog
root@LEDE:/etc/init.d# logread
- snip -
Thu Sep 14 18:26:01 2017 kern.info kernel: [ 3612.089093] wifidog[4585]: segfault at 0 ip   (null) sp bf9bee1c error 4 in wifidog[8048000+40000]
root@LEDE:/etc/init.d#
root@LEDE:/etc/init.d# cat /etc/banner
     _________
    /        /\      _    ___ ___  ___
   /  LE    /  \    | |  | __|   \| __|
  /    DE  /    \   | |__| _|| |) | _|
 /________/  LE  \  |____|___|___/|___|                      lede-project.org
 \        \   DE /
  \    LE  \    /  -----------------------------------------------------------
   \  DE    \  /    Reboot (SNAPSHOT, r4830-c88770c766)
    \________\/    -----------------------------------------------------------

root@LEDE:/etc/init.d# cat /proc/cpuinfo
processor       : 0
vendor_id       : AuthenticAMD
cpu family      : 5
model           : 10
model name      : Geode(TM) Integrated Processor by AMD PCS
stepping        : 2
cpu MHz         : 431.658
cache size      : 128 KB
fdiv_bug        : no
f00f_bug        : no
coma_bug        : no
fpu             : yes
fpu_exception   : yes
cpuid level     : 1
wp              : yes
flags           : fpu de pse tsc msr cx8 sep pge cmov clflush mmx mmxext 3dnowext 3dnow eagerfpu 3dnowprefetch vmmcall
bugs            : sysret_ss_attrs
bogomips        : 863.31
clflush size    : 32
cache_alignment : 32
address sizes   : 32 bits physical, 32 bits virtual
power management:

root@LEDE:/etc/init.d# opkg info apfree_wifidog
Package: apfree_wifidog
Version: 2.06.1395-1
Depends: libc, zlib, iptables-mod-extra, iptables-mod-ipopt, kmod-ipt-nat, iptables-mod-nat-extra, libpthread, libopenssl, libjson-c, ipset, libip4tc, libevent2, libevent2-openssl, fping, libmosquitto, libuci
Status: install user installed
Architecture: i386_pentium
Installed-Time: 1505334772

root@LEDE:/etc/init.d# cat /tmp/wifidog.conf
GatewayID apfree-test
GatewayInterface br-lan
Externalinterface eth2

AuthServer {
    Hostname xxx.xxxxxx.net
    HTTPPort 80
    Path /

}

CheckInterval 60
ClientTimeout 72000
JsFilter        1
WiredPassed     0

HTTPDMaxConn 200

PoolMode        1
ThreadNumber    5
QueueSize       20

FirewallRuleSet global {

}

FirewallRuleSet validating-users {

    FirewallRule allow to 0.0.0.0/0
}

FirewallRuleSet known-users {

    FirewallRule allow to 0.0.0.0/0
}

FirewallRuleSet auth-is-down {

}

FirewallRuleSet unknown-users {

    FirewallRule allow udp port 53
    FirewallRule allow tcp port 53
    FirewallRule allow udp port 67
    FirewallRule allow tcp port 67
}

FirewallRuleSet locked-users {

    FirewallRule block to 0.0.0.0/0
}
root@LEDE:/etc/init.d#

any idea?

liudf0716, 2017-09-14:
@KazukiShimada it's libipt's problem.

KazukiShimada, 2017-09-14:
Does it mean libipt on lede is broken?
We should wait for fix on lede?

liudf0716, 2017-09-14:
it should be our problem, we will fix it later.

closed 2017-12-04 by liudf0716