liudf0716/apfree-wifidog #174: Documentation for using certificates

johndul0001, 2017-09-30:
I hate to bother anyone but I cannot figure out how to use the certificates. What server am I supposed to certify? My auth server? The ip address of the apfree_wifidog router? I have looked and looked and there is no documentation I can find. I cannot get https portal working without knowing.

rahacker2006, 2017-10-18:
@johndul0001 same here, please help!! @liudf0716 @Akagi201 @Michael2008S 
thanks

liudf0716, 2017-10-19:
if u only want to redirect https request, using default apfree cert is ok.
if u want to apfree wifidog communicate with your auther server using https, u need to know too much,such as PKI, CA, x509, openssl blablabla, which nobody can help u but youself.

johndul0001, 2017-10-19:
AWESOME! So I notice the crt expires 2017-11-17. Everything will still work after that?

liudf0716, 2017-10-20:
I will update it

johndul0001, 2017-10-21:
That was why I asked the question in the first place. I wanted to know how to do it if you decided you didn't want to update the .crt any more. But thanks!

liudf0716, 2017-10-22:
as i answered,you question really hard to answer. anyway,thanks for you reminding.

johndul0001, 2017-11-15:
I see the latest release v2.10.1437 release still has a crt file that expires November 16, 2017. Might you please supply a new one. I will pay you.

liudf0716, 2017-11-16:
@johndul0001  no need to pay it. u can use openssl to generate your own crt. anyway, I will update it as soon as possible. thanks for your reminding.

johndul0001, 2017-11-16:
Which refers back to my first question: how?

Is this document on the right track?

https://jamielinux.com/docs/openssl-certificate-authority/introduction.html

johndul0001, 2017-11-17:
So I did a lot of work on this today using the above link. A few specific questions then.

1. apfree.ca is 2048 bit. Is this necessary or is 4096 usable?
2. must apfree.ca use your information or can we use our own?
    Yours: C = CN, ST = bj, O = kunteng.org, OU = r&d, CN = liudengfeng, emailAddress = *******@qq.com
    Mine: C = CA, ST = BC, O = lmsmarina.ca, OU = marina, CN = freewifi, emailAddress = *******@lmsmarina.ca
3. apfree.crt is 2048 bit. Is this necessary or is 4096 usable?
4. same as #2
    Yours: C = CN, ST = BJ, L = BJ, O = KUNTENG, OU = RD, CN = APFREE
    Mine: C = CA, ST = BC, L = Ladysmith, O = Marina, OU = freewifi, CN = APFREE
5. I assume CN **_MUST_** BE "APFREE"
6. If I successfully create apfree.ca, apfree.key and apfree.crt I can just replace yours in the /etc directory? Or must I do something else?

I believe this is the information some other people are looking for. I hope you will answer.

johndul0001, 2017-11-17:
Stupid me, I forgot to ask: Passwords on .ca and .crt files, yes or no?

also, for those reading these posts, the above link uses .pem files. They can just be renamed or modify the commands to create apfree.* files instead of *.pem files ....... I think :)

liudf0716, 2017-11-17:
@johndul0001 apfree certification has no different with other x509 certification, u can generate as u wanted

liudf0716, 2017-11-20:
@johndul0001 
I have update apfree certification, here u can get .
https://github.com/KunTengRom/kunteng-lede-17.01.4/tree/17.01.4/package/kunteng/net/apfree_wifidog/files

closed 2017-12-04 by liudf0716