liudf0716/apfree-wifidog #175: wifidog fail to start

deltay, 2017-10-10:
root@LEDE:/etc/config# /usr/bin/wifidog -c /tmp/wifidog.conf  -f -d 0
[ 2342.921292] do_page_fault(): sending SIGSEGV to wifidog for invalid read access from 00000000
[ 2342.929921] epc = 00000000 in wifidog[400000+28000]
[ 2342.934836] ra  = 76ca6c27 in libxtables.so.12.0.0[76ca4000+17000]
Segmentation fault

root@LEDE:/etc/config# /usr/bin/wifidog  -v
This is WiFiDog version 2.06.1395

root@LEDE:/etc/config# ldd /usr/bin/wifidog
        /lib/ld-musl-mipsel-sf.so.1 (0x77341000)
        libhttpd.so => /usr/lib/libhttpd.so (0x7732b000)
        libz.so.1 => /usr/lib/libz.so.1 (0x7730b000)
        libjson-c.so.2 => /usr/lib/libjson-c.so.2 (0x772f3000)
        libip4tc.so.0 => /usr/lib/libip4tc.so.0 (0x772de000)
        libuci.so => /lib/libuci.so (0x772c6000)
        libssl.so.1.0.0 => /usr/lib/libssl.so.1.0.0 (0x77269000)
        libcrypto.so.1.0.0 => /usr/lib/libcrypto.so.1.0.0 (0x77104000)
        libevent-2.1.so.6 => /usr/lib/libevent-2.1.so.6 (0x770bf000)
        libevent_openssl-2.1.so.6 => /usr/lib/libevent_openssl-2.1.so.6 (0x770aa000)
        libmosquitto.so.1 => /usr/lib/libmosquitto.so.1 (0x7708f000)
        libiptext.so => /usr/lib/libiptext.so (0x77071000)
        libiptext4.so => /usr/lib/libiptext4.so (0x7705b000)
        libxtables.so.12 => /usr/lib/libxtables.so.12 (0x77042000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7701f000)
        libc.so => /lib/ld-musl-mipsel-sf.so.1 (0x77341000)
        libubox.so => /lib/libubox.so (0x77006000)
        libcares.so.2 => /usr/lib/libcares.so.2 (0x76fea000)

root@LEDE:/etc/config# cat /proc/cpuinfo
system type             : Ralink RT3052 id:1 rev:3
machine                 : Asus WL-330N
processor               : 0
cpu model               : MIPS 24KEc V4.12
BogoMIPS                : 255.59
wait instruction        : yes
microsecond timers      : yes
tlb_entries             : 32
extra interrupt vector  : yes
hardware watchpoint     : yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa                     : mips1 mips2 mips32r1 mips32r2
ASEs implemented        : mips16 dsp
shadow register sets    : 1
kscratch registers      : 0
package                 : 0
core                    : 0
VCED exceptions         : not available
VCEI exceptions         : not available

deltay, 2017-10-10:
When I start /usr/bin/wifidog -c /tmp/wifidog.conf  -f -d 9,  I see some output which maybe useful :

[7][Tue Oct 10 03:23:08 2017][6163](https_common.c:50) Using OpenSSL version "OpenSSL 1.0.2l  25 May 2017"
and libevent version "2.1.8-stable"

[7][Tue Oct 10 03:23:08 2017][6163](gateway.c:498) after common_setup
[6][Tue Oct 10 03:23:08 2017][6163](gateway.c:502) Setting started_time
[7][Tue Oct 10 03:23:08 2017][6163](gateway.c:646) Finding IP address of br-lan
[7][Tue Oct 10 03:23:08 2017][6163](gateway.c:651) br-lan = 192.168.16.209
[5][Tue Oct 10 03:23:08 2017][6163](gateway.c:534) Creating web server on 192.168.16.209:2060
[7][Tue Oct 10 03:23:08 2017][6163](gateway.c:541) Assigning callbacks to web server
[6][Tue Oct 10 03:23:08 2017][6163](util.c:92) Closing ICMP socket
[7][Tue Oct 10 03:23:08 2017][6163](firewall.c:398) Removing Firewall rules
[7][Tue Oct 10 03:23:08 2017][6163](fw_iptables.c:952) Destroying our iptables entries
[7][Tue Oct 10 03:23:08 2017][6163](fw_iptables.c:965) Destroying chains in the MANGLE table
[7][Tue Oct 10 03:23:08 2017][6163](fw_iptables.c:1089) Attempting to destroy all mention of WiFiDog_br-lan_Pass from mangle.PREROUTING
[7][Tue Oct 10 03:23:09 2017][6163](gateway.c:363) Handler for SIGCHLD called. Trying to reap a child
[7][Tue Oct 10 03:23:09 2017][6163](gateway.c:369) Handler for SIGCHLD reaped child PID -1
[7][Tue Oct 10 03:23:09 2017][6163](fw_iptables.c:1089) Attempting to destroy all mention of WiFiDog_br-lan_Roam from mangle.PREROUTING
[7][Tue Oct 10 03:23:09 2017][6163](gateway.c:363) Handler for SIGCHLD called. Trying to reap a child
[7][Tue Oct 10 03:23:09 2017][6163](gateway.c:369) Handler for SIGCHLD reaped child PID -1
[7][Tue Oct 10 03:23:09 2017][6163](fw_iptables.c:1089) Attempting to destroy all mention of WiFiDog_br-lan_Trusted from mangle.PREROUTING
[7][Tue Oct 10 03:23:09 2017][6163](gateway.c:363) Handler for SIGCHLD called. Trying to reap a child
[7][Tue Oct 10 03:23:09 2017][6163](gateway.c:369) Handler for SIGCHLD reaped child PID -1
[7][Tue Oct 10 03:23:09 2017][6163](fw_iptables.c:1089) Attempting to destroy all mention of WiFiDog_br-lan_Outgoing from mangle.PREROUTING
[7][Tue Oct 10 03:23:09 2017][6163](gateway.c:363) Handler for SIGCHLD called. Trying to reap a child
[7][Tue Oct 10 03:23:09 2017][6163](gateway.c:369) Handler for SIGCHLD reaped child PID -1
[7][Tue Oct 10 03:23:09 2017][6163](fw_iptables.c:1089) Attempting to destroy all mention of WiFiDog_br-lan_Incoming from mangle.POSTROUTING
[7][Tue Oct 10 03:23:09 2017][6163](gateway.c:363) Handler for SIGCHLD called. Trying to reap a child
[7][Tue Oct 10 03:23:09 2017][6163](gateway.c:369) Handler for SIGCHLD reaped child PID -1
[7][Tue Oct 10 03:23:09 2017][6163](fw_iptables.c:259) Executing command: iptables -t mangle -F WiFiDog_br-lan_Pass
[3][Tue Oct 10 03:23:09 2017][6163](fw3_iptc.c:646) iptc_create_chain(): No chain/target/match by that name

[7][Tue Oct 10 03:23:09 2017][6163](fw_iptables.c:268) iptables command failed(0): iptables -t mangle -F WiFiDog_br-lan_Pass
[7][Tue Oct 10 03:23:09 2017][6163](fw_iptables.c:259) Executing command: iptables -t mangle -F WiFiDog_br-lan_Roam
[3][Tue Oct 10 03:23:09 2017][6163](fw3_iptc.c:646) iptc_create_chain(): No chain/target/match by that name

[7][Tue Oct 10 03:23:09 2017][6163](fw_iptables.c:268) iptables command failed(0): iptables -t mangle -F WiFiDog_br-lan_Roam
[7][Tue Oct 10 03:23:09 2017][6163](fw_iptables.c:259) Executing command: iptables -t mangle -F WiFiDog_br-lan_Trusted
[3][Tue Oct 10 03:23:09 2017][6163](fw3_iptc.c:646) iptc_create_chain(): No chain/target/match by that name

[7][Tue Oct 10 03:23:09 2017][6163](fw_iptables.c:268) iptables command failed(0): iptables -t mangle -F WiFiDog_br-lan_Trusted
[7][Tue Oct 10 03:23:09 2017][6163](fw_iptables.c:259) Executing command: iptables -t mangle -F WiFiDog_br-lan_Outgoing
[3][Tue Oct 10 03:23:09 2017][6163](fw3_iptc.c:646) iptc_create_chain(): No chain/target/match by that name

[7][Tue Oct 10 03:23:09 2017][6163](fw_iptables.c:268) iptables command failed(0): iptables -t mangle -F WiFiDog_br-lan_Outgoing
[7][Tue Oct 10 03:23:09 2017][6163](fw_iptables.c:259) Executing command: iptables -t mangle -F WiFiDog_br-lan_Incoming
[3][Tue Oct 10 03:23:09 2017][6163](fw3_iptc.c:646) iptc_create_chain(): No chain/target/match by that name

[7][Tue Oct 10 03:23:09 2017][6163](fw_iptables.c:268) iptables command failed(0): iptables -t mangle -F WiFiDog_br-lan_Incoming
[7][Tue Oct 10 03:23:09 2017][6163](fw_iptables.c:259) Executing command: iptables -t mangle -X WiFiDog_br-lan_Pass
[3][Tue Oct 10 03:23:09 2017][6163](fw3_iptc.c:697) iptc_delete_chain(): No chain/target/match by that name

[7][Tue Oct 10 03:23:09 2017][6163](fw_iptables.c:268) iptables command failed(0): iptables -t mangle -X WiFiDog_br-lan_Pass
[7][Tue Oct 10 03:23:09 2017][6163](fw_iptables.c:259) Executing command: iptables -t mangle -X WiFiDog_br-lan_Roam
[3][Tue Oct 10 03:23:09 2017][6163](fw3_iptc.c:697) iptc_delete_chain(): No chain/target/match by that name

[7][Tue Oct 10 03:23:09 2017][6163](fw_iptables.c:268) iptables command failed(0): iptables -t mangle -X WiFiDog_br-lan_Roam
[7][Tue Oct 10 03:23:09 2017][6163](fw_iptables.c:259) Executing command: iptables -t mangle -X WiFiDog_br-lan_Trusted
[3][Tue Oct 10 03:23:09 2017][6163](fw3_iptc.c:697) iptc_delete_chain(): No chain/target/match by that name

[... cut ...]
[7][Tue Oct 10 03:23:11 2017][6163](fw_iptables.c:259) Executing command: iptables -t mangle -I PREROUTING 1 -i br-lan -j WiFiDog_br-lan_Trusted
[7][Tue Oct 10 03:23:11 2017][6163](fw_iptables.c:259) Executing command: iptables -t mangle -I PREROUTING 1 -i br-lan -j WiFiDog_br-lan_Pass
[7][Tue Oct 10 03:23:11 2017][6163](fw_iptables.c:259) Executing command: iptables -t mangle -I PREROUTING 1 -i br-lan -j WiFiDog_br-lan_Roam
[7][Tue Oct 10 03:23:11 2017][6163](fw_iptables.c:259) Executing command: iptables -t mangle -A WiFiDog_br-lan_Roam -m set --match-set WiFiDog_br-lan_Roam src -j MARK --set-mark 2
Segmentation fault

liudf0716, 2017-10-11:
if u used latest LEDE release, u will encounter this problem.

closed 2017-12-04 by liudf0716

liudf0716, 2017-12-19:
pls use this sdk environment:

https://github.com/KunTengRom/kunteng-lede-17.01.4

7-cat, 2018-01-14:
I encountered the same problem, how to fix without changing the SDK environment?