liudf0716/apfree-wifidog #188: Build on Debian/Ubuntu

vsidou, 2017-11-22:
Hi,

Can we build this on a debian based linux ?
If it's possible, could you provide the dependencies please ?
As i was trying to build it, i got an error on :+1: 

apfree_wifidog-master/src/wd_util.c:54:17: fatal error: uci.h:

I'm not sure wether it'll only work on openwrt/lede (because of it's dependencies to uci) or not

thx

vsidou, 2017-11-22:
I managed to build UCI for ubuntu.
Now i get another error related to libevent2.

It's looking for a file event2/ping.h which is not found.
I assume it's a custom patch to libevent2 in order to have https redirect workable.

But i cannot find any patch to libevent2. Just prebuilt packages.
I see it use some similar code as : 
https://github.com/rcarbone/evping

But miss some functions like : 
evping_base_host_delete_by_ipname

Could you point me out to the libevent2 patch please ?

thanks

liudf0716, 2017-11-23:
@vsidou pls use release version to compile, master has some problem. If u want to try master version, pls use this libevent version:
https://github.com/KunTengRom/kunteng_libevent2

liudf0716, 2017-11-23:
@vsidou u'd better compile in this sdk environment: 
https://github.com/KunTengRom/kunteng-lede-17.01.4

liudf0716, 2017-11-24:
@vsidou 
cause apfree wifidog was aimed to use in openwrt&lede system, for example, it need to cooperate with dnsmasq, uci subsystem, while these subsystem didn't work or not necessary in Debian base system . if u only want to compile successly in Debian, all u need to do was adding library depends, such as uci, iptables, mosquitto and etc.

vsidou, 2017-11-26:
Ok i see.
Thanks for you update.
I was thinking more like making it works on a debian based embedded system like Ubiquiti Edge Router.
But it seems mostly dependent on Openwrt model.

beyondhong, 2017-12-21:
@liudf0716 I don't understand, the original wifidog is more flexible, I can port it to any platform like openwrt and ddwrt. But you make apfree only working on openwrt.

liudf0716, 2017-12-24:
@beyondhong  apfree wifidog was used in our company's product, which based on openwrt system. so we mainly focused on openwrt, that's not mean apfree wifidog can not port to other platform, in fact, some other guy has ported it  to debain and qsdk successly.

liudf0716, 2017-12-26:
can read this 
https://github.com/liudf0716/apfree_wifidog/issues/194

closed 2018-10-14 by liudf0716