liudf0716/apfree-wifidog #191: maybe keep-alive can guarantee request get right reponse?

fenggui321, 2017-12-01:
apfree_wifidog change HTTP1.0 to HTTP1.0 and support keep-alive.My question is : How apfree_wifidog assure the reponse corresponding  to exactly orignal reques? .e.g.:apfree_wifidog thread A B trigger  Request A B,both on the same socket contest ("Use keep-alive http connection"), and both thread call recv to get respnse,how to gurantee thread A---request A get reponse A ,not response B ?

liudf0716, 2017-12-03:
@fenggui321 yes, I thought what u  decription is reasonable, I cann't promise thread A's request get thread A's response, I will concel keep-alive feature before finding proper solution. thanks!

closed 2017-12-04 by liudf0716

liudf0716, 2017-12-07:
@fenggui321 did u have wechat?

fenggui321, 2017-12-07:
@liudf0716 for what ?