liudf0716/apfree-wifidog #192: How Max Concurrent Clients and Test

cuongbn, 2017-12-04:
I need to deploy the hotspot to schools with 200 people. I installed apfree_wifidog on Unifi Edgerouter Lite (512MB ram), I think this good hardware. Do not know apfree_wifidog has limited clients?
I have seen the test of apfree_wifidog , but do not know what to test concurrent client, please help me.

liudf0716, 2017-12-04:
how about your device's cpu frequency? I think apfree-wifidog can support 200 people, however, u need a good device, such as x86.

cuongbn, 2017-12-04:
This info device. https://lede-project.org/toh/hwdata/ubiquiti/ubiquiti_edgerouter_lite
Another company introduced me that they run 200 users on this device, but they use CoovaChilli
If the hardware is good enough, apfree_wifidog will not be limited Clients?

liudf0716, 2017-12-05:
@cuongbn 
apfree wifidog will never limit clients, it will never limit anything!
I can't open your url, as u know, I am  from China, we have a greate firewall. anyway, I strongly suggest u use x86 to serve 200 users.   
compared with coovachilli, apfree wifidog was more easy to use. I can't tell u more why u should choose apfree wifidog , cause I don't want to augue with anyone.

cuongbn, 2017-12-05:
Thank you, I also like to use the app base on wifidog platform like apfree_wifidog.
Also, can u help me to test the concurrent client connect gateway at the same time?

liudf0716, 2017-12-07:
we have tested apfree wifidog's performance use Webbench, u can read articles in doc directory