liudf0716/apfree-wifidog #196: Timeout device not authenticate, https, pan domains...

cuongbn, 2017-12-22:
**Timeout device not authenticate**
When I allow users to login via Facebook (I open https 443), they can access Facebook without logging in. How to force them to log in or exit of the network. I know the CheckInterval and ClientTimeout variables only apply to logged in users (I run / usr / bin / wifidog -c /tmp/wifidog.conf -f -d 9 so I can see the log)

**https**
When I open the https 443 port
> FirewallRuleSet unknown-users {
>       FirewallRule allow tcp port 443
     => Error, unregistered user can not access https
> FirewallRuleSet global {
>       FirewallRule allow tcp port 443
    => Done, unregistered users can access https (facebook, google ...)

**Pan Domains**
The whole system runs fine, but when I add the settings
> TrustedPanDomains facebook.com, facebook.net, fbcdn.net
Wifidog run error, i track log by run / usr / bin / wifidog -c /tmp/wifidog.conf -f -d 9.  [https://pastebin.com/WywL3tKV ](url)

liudf0716, 2017-12-22:
@cuongbn  u must be sure to compile dnsmasq with ipset support.

cuongbn, 2017-12-23:
@liudf0716  How kick out device user after sometime or reopen captive page on them device

closed 2018-10-14 by liudf0716