liudf0716/apfree-wifidog #199: problem when ip host change sever not ping or auth token to sever.

tidus30691nexus5, 2017-12-27:
Am using last version.
Host using sevice cloudflare.

When first all runnning ok and ip auth sever is: 104.22.6.100 . But when host sever ( use cloudflare ) change ip router not ping or auth to my sever . After am restart wifidog it work again and am seen ip auth sever change (104.22.5.100)

am think have problem on function _connect_auth_server 

please help me check problem

liudf0716, 2017-12-27:
@tidus30691nexus5 next release we will fix this problem.

tidus30691nexus5, 2017-12-27:
am think problem is firewall not add new ip when host server auth change new ip. 
because wifidog till running. am can use wdctl status to see client online and active

liudf0716, 2017-12-28:
yes, wdctl status can get what u want.

tidus30691nexus5, 2017-12-28:
hi @liudf0716 ,
Am add more config **list firewall_rule_global	'allow to 104.22.0.0/22**' and run again,
After run 6hour wifidog not ping or auth to sever auth again.
But am check wdctl status it till running on router. So when unknow user connect wifi can't show page captive for user login.
Please help me fix it. Thanks

tidus30691nexus5, 2017-12-28:
@liudf0716 
problem is when ip change it mark server down and never recheck again .

closed 2018-12-20 by liudf0716