liudf0716/apfree-wifidog #213: target "CT" has version "libxtables.so.12", but "libxtables.so.10" is required

jahmed786, 2018-10-19:
I built apfree_wifidog on ubuntu 14 for my target platform using the kunteng-lede-17.01.4 enviorment from git repo.
I installed the ipk package on OpenWrt 18.06.1 and copied libevent-2.1.so.6, libevent_openssl-2.1.so.6 and libxtables.so.10 from the compiled project to /usr/lib/ and created soft links for them in /usr/lib64/. 
Upon running /usr/bin/wifidog what I get is:
[6][Fri Oct 19 13:20:03 2018][3842](conf.c:973) Reading configuration file '/etc/wifidog.conf'
[6][Fri Oct 19 13:20:03 2018][3842](conf.c:1880) Locking domain
[6][Fri Oct 19 13:20:03 2018][3842](conf.c:1880) Domains locked
[6][Fri Oct 19 13:20:03 2018][3842](gateway.c:907) Forking into background
root@OpenWrt:~# [6][Fri Oct 19 13:20:03 2018][3843](gateway.c:512) Setting started_time
[5][Fri Oct 19 13:20:03 2018][3843](gateway.c:544) Creating web server on 192.168.20.10:2060
[6][Fri Oct 19 13:20:03 2018][3843](util.c:93) Closing ICMP socket
(null): target "CT" has version "libxtables.so.12", but "libxtables.so.10" is required.
I copied libxtables.so.10 from this path
lede/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_mt7628/iptables-1.4.21/ipkg-mipsel_24kc/libxtables/usr/lib

What can I do to fix this ?

liudf0716, 2018-10-20:
u means u want to run apfree wifidog in x86 ubuntu  by copying mips's libevent lib to ubuntu ?

jahmed786, 2018-10-24:
I want to run apfree wifidog on my TP-Link (TL-MR3420 Version 5) router. It is running OpenWrt 18.06.1.

I am building it from the source on ubuntu and I get the above error when I try to install it.

jahmed786, 2018-10-27:
I had to replace  the libevent, libevent_openssl, libiptext.so, libiptext4.so and libiptext6.so files to get rid of these error messages.

closed 2018-10-27 by jahmed786