liudf0716/apfree-wifidog #217: apfree_wifidog doesn't run on gl-inet b1300 router (Set WiFiDog_br-lan_Roam doesn't exist.)

nuwanarti, 2018-12-18:
[7][Tue Dec 18 07:31:33 2018][5886](fw_iptables.c:260) Executing command: iptables -t mangle -I PREROUTING 1 -i br-lan -j WiFiDog_br-lan_Roam
[7][Tue Dec 18 07:31:33 2018][5886](fw_iptables.c:260) Executing command: iptables -t mangle -A WiFiDog_br-lan_Roam -m set --match-set WiFiDog_br-lan_Roam src -j MARK --set-mark 0x020000/0xff0000
(null) v4: Set WiFiDog_br-lan_Roam doesn't exist.

nuwanarti, 2018-12-18:
full debug out of wifidog 

root@GL-B1300:~/base# /usr/bin/wifidog -c /tmp/wifidog.conf -f -d 8
[6][Tue Dec 18 07:31:32 2018][5886](conf.c:869) Reading configuration file '/tmp/wifidog.conf'
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:915) Parsing token: GatewayID, value: E4956E4510B3
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:915) Parsing token: GatewayInterface, value: br-lan
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:915) Parsing token: Externalinterface, value: eth0
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:915) Parsing token: AuthServer, value: {
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:514) Adding ec2-52-204-51-163.compute-1.amazonaws.com:3000 (SSL: 443) / to the auth server list
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:542) Auth server added
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:915) Parsing token: CheckInterval, value: 60
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:915) Parsing token: ClientTimeout, value: 72000
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:915) Parsing token: JsFilter, value: 1
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:915) Parsing token: WiredPassed, value: 0
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:915) Parsing token: HTTPDMaxConn, value: 200
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:915) Parsing token: PoolMode, value: 1
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:915) Parsing token: ThreadNumber, value: 5
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:915) Parsing token: QueueSize, value: 20
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:915) Parsing token: FirewallRuleSet, value: global
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:657) Adding Firewall Rule Set global
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:710) Firewall Rule Set global added.
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:915) Parsing token: FirewallRuleSet, value: validating-users
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:657) Adding Firewall Rule Set validating-users
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:693) p1 = [FirewallRule]; p2 = [allow to 0.0.0.0/0]
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:734) leftover: allow to 0.0.0.0/0
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:810) Adding Firewall Rule allow (null) port (null) to 0.0.0.0/0
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:710) Firewall Rule Set validating-users added.
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:915) Parsing token: FirewallRuleSet, value: known-users
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:657) Adding Firewall Rule Set known-users
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:693) p1 = [FirewallRule]; p2 = [allow to 0.0.0.0/0]
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:734) leftover: allow to 0.0.0.0/0
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:810) Adding Firewall Rule allow (null) port (null) to 0.0.0.0/0
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:710) Firewall Rule Set known-users added.
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:915) Parsing token: FirewallRuleSet, value: auth-is-down
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:657) Adding Firewall Rule Set auth-is-down
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:710) Firewall Rule Set auth-is-down added.
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:915) Parsing token: FirewallRuleSet, value: unknown-users
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:657) Adding Firewall Rule Set unknown-users
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:693) p1 = [FirewallRule]; p2 = [allow udp port 53]
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:734) leftover: allow udp port 53
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:810) Adding Firewall Rule allow udp port 53 to 0.0.0.0/0
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:693) p1 = [FirewallRule]; p2 = [allow tcp port 53]
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:734) leftover: allow tcp port 53
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:810) Adding Firewall Rule allow tcp port 53 to 0.0.0.0/0
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:693) p1 = [FirewallRule]; p2 = [allow udp port 67]
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:734) leftover: allow udp port 67
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:810) Adding Firewall Rule allow udp port 67 to 0.0.0.0/0
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:693) p1 = [FirewallRule]; p2 = [allow tcp port 67]
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:734) leftover: allow tcp port 67
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:810) Adding Firewall Rule allow tcp port 67 to 0.0.0.0/0
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:710) Firewall Rule Set unknown-users added.
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:915) Parsing token: FirewallRuleSet, value: locked-users
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:657) Adding Firewall Rule Set locked-users
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:693) p1 = [FirewallRule]; p2 = [block to 0.0.0.0/0]
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:734) leftover: block to 0.0.0.0/0
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:810) Adding Firewall Rule block (null) port (null) to 0.0.0.0/0
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:710) Firewall Rule Set locked-users added.
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:1766) Parsing string [www.kunteng.org,cloud.kunteng.org,mqtt.kunteng.org,log1.kunteng.org] for trust domains
[6][Tue Dec 18 07:31:32 2018][5886](conf.c:1772) Locking domain
[6][Tue Dec 18 07:31:32 2018][5886](conf.c:1772) Domains locked
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:1791) Adding&Delete [1] trust domain [www.kunteng.org] to&from list
[7][Tue Dec 18 07:31:32 2018][5886](conf.c:1791) Adding&Delete [1] trust domain [cloud.kunteng.org] to&from list
[... cut ...]
[7][Tue Dec 18 07:31:33 2018][5886](fw_iptables.c:260) Executing command: iptables -t mangle -I PREROUTING 1 -i br-lan -j WiFiDog_br-lan_Pass
[7][Tue Dec 18 07:31:33 2018][5886](fw_iptables.c:260) Executing command: iptables -t mangle -I PREROUTING 1 -i br-lan -j WiFiDog_br-lan_Roam
[7][Tue Dec 18 07:31:33 2018][5886](fw_iptables.c:260) Executing command: iptables -t mangle -A WiFiDog_br-lan_Roam -m set --match-set WiFiDog_br-lan_Roam src -j MARK --set-mark 0x020000/0xff0000
(null) v4: Set WiFiDog_br-lan_Roam doesn't exist.

liudf0716, 2018-12-18:
@nuwanarti  execute 'ipset list', can u find WiFiDog_br-lan_Roam; if not, I think it's ipset's problem.

nuwanarti, 2018-12-18:
root@GL-B1300:~/base# ipset list
Name: mwan3_connected
Type: hash:net
Revision: 4
Header: family inet hashsize 1024 maxelem 65536
Size in memory: 8952
References: 2
Members:
127.255.255.255
127.0.0.0/8
192.168.8.0
192.168.8.1
127.0.0.0
224.0.0.0/3
192.168.1.1
127.0.0.1
192.168.1.255
192.168.1.0
192.168.8.0/24
192.168.1.0/24
192.168.1.34
192.168.8.255

Name: mwan3_track_wan
Type: hash:ip
Revision: 2
Header: family inet hashsize 1024 maxelem 65536
Size in memory: 8312
References: 1
Members:
208.67.220.220
208.67.222.222
8.8.8.8
8.8.4.4

Name: WiFiDog_br-lan_TDomains
Type: hash:ip
Revision: 2
Header: family inet hashsize 1024 maxelem 65536
Size in memory: 8248
References: 0
Members:

Name: WiFiDog_br-lan_ITDomains
Type: hash:ip
Revision: 2
Header: family inet hashsize 1024 maxelem 65536
Size in memory: 8248
References: 0
Members:

Name: WiFiDog_IPSET_TDomains
Type: hash:ip
Revision: 2
Header: family inet hashsize 1024 maxelem 65536
Size in memory: 8248
References: 0
Members:

nuwanarti, 2018-12-18:
I have ipset version 6.24 installed on the router. does apfree_wifidog supports ipset v 6.24 if isn't where would I find the source of ipset v6.3 to compile for openwrt

liudf0716, 2018-12-18:
apfree wifidog use ipset command to create hash:mac SETNAME, for lower version ipset, which didn't support that feature. u can exucute `ipset -h|grep mac` to find whether your ipset support that feature.

nuwanarti, 2018-12-18:
this is the output
root@GL-B1300:~# ipset -h|grep mac
    hash:mac		0	Initial revision
    hash:ip,mac		0	Initial revision
    bitmap:ip,mac	3	skbinfo support
    bitmap:ip,mac	2	comment support
    bitmap:ip,mac	1	counters support
    bitmap:ip,mac	0	Initial revision

nuwanarti, 2018-12-18:
I compiled and installed ipset v 7 but the error stays  the same

liudf0716, 2018-12-18:
which version u used?

nuwanarti, 2018-12-18:
in gl-inet ar 300m router apfree_wifidog is working fine. ipset version in that router is v6.30. In gl-inet b1300 router apfree_wifidog isn't working. ipset version is v6.24. so I recompiled newer version(v7.0) of ipset for b1300 router and installed. but the same error occured

liudf0716, 2018-12-18:
which apfree wifidog version u used?
u can use `wifidogx -c /tmp/wifidog.conf -f -d 7` to watch whether that problem still exist

nuwanarti, 2018-12-18:
WiFiDog version 3.03.1480
apfree wifidog https://github.com/liudf0716/apfree_wifidog/tree/master

in gl-inet b1300 router shipped with ipset v6.24 installed. so kernel support is available. 
and i couldn't understand the command 

_ipset create yourname  hash:mac timeout 0_

nuwanarti, 2018-12-18:
its the very first command I used .. and its given the above mentioned error

liudf0716, 2018-12-18:
@nuwanarti  if u router's kernel didn't support ipset's 'hash:mac' feature, only updating ipset's userspace program was useless. 
your problem is the kernel version is lower.

closed 2018-12-20 by liudf0716

nuwanarti, 2018-12-20:
Thank you for your help