liudf0716/apfree-wifidog #221: auth server offline

khoailangdo, 2019-06-27:
I have configured in /etc/config/wifidogx like below:
option auth_server_hostname 'dattest.matmagoc.com'
option auth_server_port 443
option auth_server_path '/signin/'

After i start with command success:  /etc/init.d/wifidogx start
ApFree-wifidog redirect to Auth Server Offline page?

What happened. Or do I configure something wrong? Help me handle this problem
Thanks

liudf0716, 2019-06-29:
does your auth server support https? if not, pls change auth_server_port from 443 to your auth server port

khoailangdo, 2019-06-30:
ok thanks. I have already fixed it.
I have already fixed it.
I have a new problem, about allowing a domain to whitelist:
`option trusted_domains 'connectivitycheck.android.com, connectivitycheck.gstatic.com, lichngaytot.com'`
When I have not authenticated, I only point Ping and access the domain lichngaytot.com, while other domains I cannot Ping.  Are there any errors in the configuration?

liudf0716, 2019-07-06:
u can try `wdctx show_trusted_domains` to see whether the rule applied.

closed 2019-10-18 by khoailangdo