liudf0716/apfree-wifidog #223: Pls, comments/docs in English

reinerotto, 2019-08-06:
As an intensive user of coova-chilli, I am looking for an alternative, because of performance issues. 
To compare coovas functions, and apfree_wifidogs functionality, and as a general courtesy to the community, it would be very helpful to have complete English comments or docs. 
googles translate is not very helpful :-)

liudf0716, 2019-08-07:
thanks for your advise, I will try my best to provide more english document.

superliu666, 2019-08-07:
Hi I am responsible to translate this document and if you have any question or problem please chat with me, 🙏 thanks

superliu666, 2022-05-16:
这是来自QQ邮箱的假期自动回复邮件。你好，我最近正在休假中，无法亲自回复你的邮件。我将在假期结束后，尽快给你回复。

SohailAsghar236, 2024-05-14:
> Hi I am responsible to translate this document and if you have any question or problem please chat with me, 🙏 thanks

Is there a way to translate the app into English? Or an English language setting is available?

liudf0716, 2024-05-15:
> > Hi I am responsible to translate this document and if you have any question or problem please chat with me, 🙏 thanks
> 
> Is there a way to translate the app into English? Or an English language setting is available?

English language setting is default available.

closed 2024-11-14 by liudf0716