liudf0716/apfree-wifidog #224: Send Auth data null to server

khoailangdo, 2019-10-18:
I have problems sending Auth data to the server. When I started to install Apfree-Wifidog, the system worked fine. Auth data submitted successfully and fully.
But after 1 month I checked the log on the server and found that their values are null as:
stage = counters_V2, gw_id = null, query = null,ip= null, mac= null ,....
Is the error in the source code of apfree-wifidog or some service stopped?
Thanks

liudf0716, 2019-10-21:
@khoailangdo which version？

khoailangdo, 2019-10-23:
I am using apfree-wifidog-3.11.1716 version
I fixed the above error, but now I get a new error:
- With an Android device, the normal captive portal is displayed. The device must be logged in to access the internet.
- But with iOS device, you do not need to login with captive portal to still access the internet.
Why is it with such an iOS device. Did I not turn on any function?

closed 2019-10-23 by khoailangdo