liudf0716/apfree-wifidog #240: support nft

liudf0716, 2023-01-31:


liudf0716, 2023-02-02:
chatgpt gives the following code to implement the replacement of variables:

```
#include <stdio.h>
#include <string.h>

char* replace_variables(const char* auth_server_ip, const char* gateway_ip, const char* interface, const char* filename) {
    FILE* input_file;
    FILE* output_file;
    char line[1024];
    char buf[1024];
    char output_filename[1024];
    int len;
    
    // Open the input file for reading
    input_file = fopen(filename, "r");
    if (input_file == NULL) {
        return NULL;
    }
    
    // Create the output file name by appending ".out" to the input file name
    strcpy(output_filename, filename);
    strcat(output_filename, ".out");
    
    // Open the output file for writing
    output_file = fopen(output_filename, "w");
    if (output_file == NULL) {
        fclose(input_file);
        return NULL;
    }
    
    // Read the input file line by line
    while (fgets(line, sizeof(line), input_file) != NULL) {
        // Replace $auth_server_ip$ with auth_server_ip
        strcpy(buf, line);
        len = strlen(buf);
        for (int i = 0; i < len - 13; i++) {
            if (strncmp(buf + i, "$auth_server_ip$", 15) == 0) {
                strcpy(buf + i, auth_server_ip);
                strcpy(buf + i + strlen(auth_server_ip), line + i + 15);
                break;
            }
        }
        
        // Replace $gateway_ip$ with gateway_ip
        len = strlen(buf);
        for (int i = 0; i < len - 11; i++) {
            if (strncmp(buf + i, "$gateway_ip$", 13) == 0) {
                strcpy(buf + i, gateway_ip);
                strcpy(buf + i + strlen(gateway_ip), line + i + 13);
                break;
            }
        }
        
        // Replace $interface$ with interface
        len = strlen(buf);
        for (int i = 0; i < len - 9; i++) {
            if (strncmp(buf + i, "$interface$", 11) == 0) {
                strcpy(buf + i, interface);
                strcpy(buf + i + strlen(interface), line + i + 11);
                break;
            }
        }
        
        // Write the replaced line to the output file
        fputs(buf, output_file);
    }
    
    // Close the input and output files
    fclose(input_file);
    fclose(output_file);
    
    // Return the output file

```

liudf0716, 2023-02-03:
Before we run the NFT rule file of apfree-wifidog, we should delete all the OpenWrt firewall rules, the following code should do the job.

```
system("/etc/init.d/firewall restart");
system("nft -f wifidogx");
```

helintongh, 2023-02-03:
> Before we run the NFT rule file of apfree-wifidog, we should delete all the OpenWrt firewall rules, the following code should do the job.
> 
> ```
> system("/etc/init.d/firewall restart");
> system("nft -f wifidogx");
> `

helintongh, 2023-02-03:
In fw4 fw_allow must excute 
```
add rule inet fw4 mangle_prerouting_wifidogx_outgoing ether saddr $smac$ ip saddr $saddr$ counter mark set 0x20000 accept
add rule inet fw4 mangle_postrouting_wifidogx_incoming ip daddr $saddr$ counter accept
```
Is it must run delete command before add?

liudf0716, 2023-02-03:
@helintongh when adding an authenticated client, i think there is no need to check whether the client has been added to the chain. however, when deleting a client, we should check whether the client has been completely deleted, as the client may have been added twice or more.

liudf0716, 2023-05-03:
I have finished the feature.

closed 2023-05-03 by liudf0716