liudf0716/apfree-wifidog #244: 4.08.1771 存在内存泄露呢

zhoujianwen01, 2023-06-25:
1.每当设备进行认证后，内存会存在增长，超时后重复认证也会增加，且设备断开连接后也不会释放
2.当设备接入，且不进行认证操作，内存也会存在增长

liudf0716, 2023-06-26:
@zhoujianwen01 谢谢反馈，后面有空我会跟进。

liudf0716, 2023-06-26:
最新版本改动较大，可以看看是否有同样的现象。

alone-art, 2023-06-26:
@zhoujianwen01 开了什么功能, 我最近也在用这个版本, 我也测测我的

zhoujianwen01, 2023-06-26:
> @zhoujianwen01 开了什么功能, 我最近也在用这个版本, 我也测测我的

配置是默认的，没有更改

zhoujianwen01, 2023-07-06:
> 最新版本改动较大，可以看看是否有同样的现象。

hello，这个是由于libevent没有使用内存池，频繁申请释放小块内存，存在太多内存碎片导致，需要在合适的地方调用malloc_trim(0)将内存归还给操作系统

liudf0716, 2023-07-07:
@zhoujianwen01 能提个PR解决这个问题吗？

zhoujianwen01, 2023-07-10:
> @zhoujianwen01 能提个PR解决这个问题吗？

我挂测后发现并不能完全解决这个泄露问题，目前我们是在线程里面定时执行malloc_trim(0)回收资源，但挂测两天后，这个泄露问题依然存在

liudf0716, 2023-07-18:
@zhoujianwen01 @alone-art 最新的代码找到几处内存泄漏的地方，已经修改提交了，另外优化了一处处理，可以测试验证一下是否缓解内存泄漏的情况。

zhoujianwen01, 2023-07-19:
> @zhoujianwen01 @alone-art 最新的代码找到几处内存泄漏的地方，已经修改提交了，另外优化了一处处理，可以测试验证一下是否缓解内存泄漏的情况。

我们更新了libevent的库，其中包含了你的这几处改动，挂测的话内存泄露有缓解，但是仍然存在泄露

zhoujianwen01, 2023-07-19:
> > @zhoujianwen01 @alone-art 最新的代码找到几处内存泄漏的地方，已经修改提交了，另外优化了一处处理，可以测试验证一下是否缓解内存泄漏的情况。
> 
> 我们更新了libevent的库，其中包含了你的这几处改动，挂测的话内存泄露有缓解，但是仍然存在泄露

并且我们配合了定时使用了malloc_trim(0)

liudf0716, 2023-07-19:
@zhoujianwen01 好的，谢谢反馈，我再review一下代码。

liudf0716, 2023-07-21:
@zhoujianwen01 @alone-art 更新了代码，又找了几次内存泄漏点。有空可以验证下，另外malloc_trim是必须的吗？

closed 2023-07-26 by liudf0716